nvdla/hw #3: compile error(need DW02_tree.v)

rill-zhen, 2017-09-26:
when i run "make" in the verif/sim directory，i got the error info. as follow:
"xxxx/xxxx/DW02_ree.v"
cannot be oped or reading due to "No such file or directory"

rill-zhen, 2017-09-26:
i tried to modify it to "DESIGNWARE_NOEXIST ?=1", it dose not work.

rill-zhen, 2017-09-26:
it worked when i add "+define+DESIGNWARE_NOEXIST" to vcs cmd line.

jwise, 2017-09-26:
Yeah, `DESIGNWARE_NOEXIST` is both an environment variable and a Verilog macro.  The environment variable should also probably set the Verilog macro; I'll leave this issue open to track that.  Thanks for the bug report!

zhangpengsjtu, 2017-09-27:
Hi, zdraw,
     I think we should set DESIGNWARE_NOEXIST 1 by default, as we assume DW* libs are copied under vmod/vlibs even when user has DW*

jwise, 2017-09-27:
Dan and @zdraw and I were talking about this earlier today.  The problem is that for synthesis applications that have DW, this will result in synthesis and simulation not coming from the same RTL, which would be bad.  We'd really rather people use DW components, I think.

jwise, 2017-09-27:
(Leaving this issue open for me to update the docs tomorrow morning, too.)

nvdsmith, 2017-09-27:
The reason DESIGNWARE_NOEXIST is 0 by default is because QOR is better with the DW* cells.  I expect this will be addressed at some point soon.

closed 2017-09-28 by jwise

redpanda3, 2018-04-28:
Where is the environment variable and verilog macro location? In spec/def?

nvdsmith, 2018-04-30:
In the latest release (master branch) the variable should be set in your tree.make file.