nvdla/hw #5: Syntax/semantic isssue in RANDFUNC hash gen code

svenka3, 2017-09-26:
Hello,
  First of all great initiative! Thanks!

I am trying to  compile/run this code on Mentor's Questa. I found one small issue (As reported by the tool). 

File:  vmod/vlibs/RANDFUNC.vlib

There is a hash generator function. it uses an assignment like below:

  modName = $sformatf ("%m");

With modName declared as "reg" it doesn't accept - as string to a packed type requires a cast. I did a simple fix as in:

	modName = int'($sformatf ("%m"));

Not sure if that's enough as the modName could be larger than 32 bits maybe? But it maks the compiler happy and it proceeds to running the test (Still fighting other issues in that, so no results yet).

I believe Questa is right in this case and you may be able to run the code with the above change in VCS as well. if not we could add a `ifdef VCS stuff.

Thanks for listening.

Regards
Srini

svenka3, 2017-09-26:
BTW that change is needed in 2 places inside the same file.

jwise, 2017-09-26:
Thanks @svenka3.  @zdraw, can you see if this works ok with VCS?

zdraw, 2017-09-26:
Works in VCS, @svenka3 please reopen if it's still an issue.

svenka3, 2017-11-19:
This issue seems to be back in a recent build. Kindly fix.

Thanks
Srini

jwise, 2017-11-20:
Yep, confirmed.  Sorry about that.  I'm taking a look now.

jwise, 2017-11-27:
I've fixed this in the nvdlav1 branch.

commit 43ae5e3c (2017-09-26): Cast modName assignments as an int since it's declared as reg. VCS shows no warnings either way but it fixes Questa compile. Fixes #5
closed 2017-11-27 by jwise