nvdla/hw #8: Makefile issues

korenchkyn, 2017-09-27:
Could you please explain what does the "make verifcom DUMP=1 DUMPER=VERDI" stands in the simulation makefile? I cannot find any reference to "verifcom".

Also if I use make run, make verdi, verdi (as stated in the example) cannot find "simvwork" library and debussy.fsdb. Is it a makefile related error, or am I missing a step?

zdraw, 2017-09-27:
It's supposed to be vericom but muscle memory kicked in when writing that comment.
"make build DUMP=1 DUMPER=VERDI" will setup VERDI_PLI and add them to the initial simv compile. This step is required for dumping waves. 
"make vericom DUMP=1 DUMPER=VERDI" will build the simvwork library so that the design is loaded in verdi and you can correlate signals to the source code
"make run DUMP=1 DUMPER=VERDI" will run the test and dump the debussy.fsdb file
"make verdi DUMP=1 DUMPER=VERDI" will load the library and fsdb file.

korenchkyn, 2017-09-27:
Oh that makes sense, thank you very much. I've grepped for verifcom, haven't found anything and thought it might be another NV specific inside tool or so. Guess I will have a good night, playing with the HDL code. ^^
Thank you once again!

closed 2017-09-27 by korenchkyn

jwise, 2017-09-27:
I updated the Makefile comment.