nvdla/hw #9: Verilog - input declared as reg?

svenka3, 2017-09-27:
File: synth_tb/csb_master_seq.v 
Lines:   23 (approx):

    input reg [31:0] mcsb2mseq_rdata;
    input reg dut2mseq_intr0;

Is that right? Somehow other tools are fine, but Icarus Verilog chokes on this. 

Regards
Srini

Wyp184800, 2017-09-28:
This is a testbench file. 
Do not synthesis any testbench.

zdraw, 2017-09-28:
For Issue #6 and issue #7 and this one, I'm seeing if I can bring up Icarus.

svenka3, 2017-09-29:
Please let me know if you need help with Icarus script and changes

svenka3, 2017-09-30:
You can find a new dir named run_dir to run on different tools. 

https://github.com/svenka3/NVDLA_hw

cd verif
cd run_dir

make cvc5 --> Compile with Icarus

Changes are done in few files and marked with "VW changed for Icarus" in comments.

Regards
Srini

zdraw, 2017-10-02:
Thanks, using your filelist I'm attempting to build the project using Icarus. We're doing a small release on Wednesday and I'll fix these compile issues in there. The long build time though...

jwise, 2017-10-07:
Also fixed in 5d6de06.  Thanks!

closed 2017-10-07 by jwise