nvdla/hw #12: support ROI pooling?

apolonic, 2017-09-28:
could you enlight me about how the ROI pooling is supported?

pengpeterli, 2017-10-07:
NVDLA is always doing "ROI" operations. NVDLA is always processing part of feature maps in a larger feature maps.

Convolution is actually ROI convolution, pooling is actually ROI pooling.

In conception, for input feature map configuration, ROI width and whole width are both required. For height, ROI height and width*height are both required. Actually configuration is a little bit different due to NVDLA specified data structure and hardware resource saving.

Configuration details will be revealed in later released documents.

HKLee2040, 2017-11-21:
Does NV have faster rcnn example that can be deployed on NVDLA?