nvdla/hw #14: support "locally connect conv"?

apolonic, 2017-09-28:
It means the weight of the convolution is loacal and not shared.
Could you enlight me how it could be supported?

nvdsmith, 2017-09-29:
This isn't one of the hardware layers targeted by the NVDLA core. Locally connected convolution can be mapped to the NVDLA, but given the large number of weights needed, bandwidth would limit MAC utilization.  This might be fine in an IoT device depending on the network size.  Is this something that's important to your application?

nvdsmith, 2017-09-29:
Also, with batching I'd expect MAC utilization would be good with current weight bandwidth.  Will need to map this out in more detail at some point though if it's important.

apolonic, 2017-09-30:
@nvdsmith  Thanks for the reply, no, I don't have to stick on this feature. There is plenty of much features to select. I see your point : the tool chain can map it into the NVDLA. 

Actually it is by my interested to know how much it can do. Any pic or talking of structure and data format will be more helpful. Thanks.

peng-peter-li, 2017-10-09:
Actually, for each kernel, it operates with a feature map which has the same width and height as kernel's. It could be considered as a special case of fully connected layer.

Current optimization for fully connected layer is suggesting users to batch input feature maps.

In current version, for locally connected layer, if convolution stride is less than kernel size, there could be re-fetching penalty on input feature map.

We will release more documents - including the data format you were mentioned - later.

HKLee2040, 2017-11-14:
Locally connected layer is useful to facial recognition.

closed 2017-11-17 by apolonic