nvdla/hw #16: common perl path

apolonic, 2017-09-28:
#! /usr/bin/env perl
this way could be better than 
#!/home/utils/perl-5.8.8/bin/perl
in :
hw/verif/synth_tb/sim_scripts/checktest_synthtb.pl
hw/verif/synth_tb/sim_scripts/raw_mem_to_synth_mem.pl
...

galv, 2017-09-28:
Similar comments apply to `hw/verif/sim/Makefile`, though not specifically about  There appear to be quite a few hardcoded paths in there, including a non-existent `./bin2bmp.pl`

It strikes me that some sloppiness such that it's hard to get a one-click install is rather the norm in hardware projects. Hopefully the open-source nature of the project will help clean this up over time.

jwise, 2017-09-29:
Thanks for the feedback -- we're tracking this and will see if we can start pulling some of the hardcoded paths out.  As @galv correctly noted, the way hardware projects are built is quite different from the way software projects are built!  We do aspire to build not just a high-quality design, but high-quality infrastructure, so if you see more things like this, please feel free to file issues for them.

jwise, 2017-10-07:
Ok, we've gone for a bunch of these in 84b85c2.  @galv, if you have other specific instances of this kind of issue, can you open a bug for them (or reopen this one)?  We're working on coming up with a revamped build flow, too, that should be generally be a little more friendly to configure; if there are specific paths that have escaped the sample config scripts, though, please continue to point them out and we'll do the best we can to clean them up.

closed 2017-10-07 by jwise