nvdla/hw #17: performance calculator not update after modifying "Configuration Input"

zhao123yongke, 2017-09-28:
How to   use the sheet | The   "Configuration Input" sheet exposes the hardware parameters for the   configuration of interest.  Fill in   your configuration settings and view the resulting performance in the   "Performance Sumary" sheet.    This sheet supports several very common CNN Networks. Additional   sheets can be added in a similar fashon to estimate performance on other   convolutional networks.
-- | --

For example, I changed "Clock   Frequency (MHz)" from 1000MHz to 2000MHz, in theory it should half the runtime or double the FPS. But actually nothing happend.

ZackyZhou, 2017-10-01:
Could you check whether the layers of your interesting network is "DRAM" limited?  If that is the case, then increasing the clock frequency (i.e. increase the calculation resource) won't get improve the final performance.

ZackyZhou, 2017-11-14:
@zhao123yongke Could you confirm whether this issue is already resolved or not?

closed 2017-12-07 by jwise