nvdla/hw #19: how to get different configuration of rtl .

yzt000000, 2017-09-30:


jwise, 2017-10-01:
We're still working on massaging the RTL to make the configuration parameters accessible.  We hope that in Q4 we'll have a "small configuration" early access release available for you; you can keep track of our plans [on our roadmap](http://nvdla.org/roadmap.html).

xalogic-linus, 2017-11-15:
Hi @jwise 

Since the parsing script allows both nv_large and nv_small, should I assume the nv_small is available?

The reason I ask is that I did a trial synthesis and found both size a little different but not significant, something like 2-5%. Just trying to figure out if I am doing anything wrong.

Also can NVDIA share the estimated Large vs Small gate count?

Linus

jwise, 2017-11-15:
Although that script has a little bit of early support for small vs. large, the small release isn't really ready yet (and is not fully minimized).  We'll plan to branch what we currently have -- the "nv_large" config -- into its own branch, and soon, we'll have in the master branch a more configurable release.  (The configurable release will be int8-only for the initial release, which will additionally save area that way.)

I think we have area estimates for the large release in 16nm and 28nm; I recognize that gate count plus ram size is more representative for FPGA implementations, though.

ratnayak, 2018-01-16:
we are having the same issue as Linus (above). we have synthesized nvdla_small and the area/gate count is not significantly different than nvdla_full. 
we like to have a scaled down INT8 version. hence just like to check whether nvdla_small is ready or when would it be ready. 

Thanks
Ruwan

jwise, 2018-01-16:
As mentioned above, nv_small is not ready yet.  We anticipate releasing a build that will support nv_small in the near future.

ratnayak, 2018-01-17:
Current nv_full indicates it has 2048 MACs? Does it mean 2048 floating-point-MACs or 1024 floating-point-MACs? Thanks.

ratnayak, 2018-01-24:
If possible can we can get some idea on the timeline for nv_small--this will greatly help us in order for us to make some decisions. What is the expected release dates--e.g. in 1/3/6 months?
Thanks.

nvhook, 2018-07-03:
The nv_small configuration is available in the master branch. 
Closing issue.

closed 2018-07-03 by nvhook