nvdla/hw #20: Regarding testbench simulation

sethu018, 2017-10-02:
I tried to run sanity test using,
   **make sanity**
   The test hangs. **make build** compiled successfully though. 
   The screenshot of the terminal screen after running make sanity is shown below:
[image]

zdraw, 2017-10-02:
Hi, the inp_txn_to_hexdump.pl script is erroring out because we specified #!/home/utils/perl-5/8/8/bin/perl.  Try changing that to point to your perl location. I'll keep this open until we make a nicer error message and change the perl header.

sethu018, 2017-10-02:
Hi, The sanity test ran after making the suggested changes. Thank you.

closed 2017-10-02 by sethu018