nvdla/hw #21: Synthesis Error

sethu018, 2017-10-06:
Hi,

 I am seeing the following error while running **syn/scripts/syn_launch.sh** script.

_######################
## Section : synthesis
#####################
setVar RTL_DEPS ""
Info: Setting RTL_DEPS from env, value = osdla_syn_20171004_1853/scripts/NV_NVDLA_partition_a.files.vc
set vcsOpt "{-f $RTL_DEPS}"
{-f osdla_syn_20171004_1853/scripts/NV_NVDLA_partition_a.files.vc}
catch {eval {analyze -format sverilog -work WORK} -vcs $vcsOpt} analyzeStatus
Running PRESTO HDLC
Searching for ./NV_NVDLA_partition_a.v
Searching for osdla_syn_20171004_1853/src/NV_NVDLA_partition_a.v
Error:  Unable to open file `NV_NVDLA_partition_a.v': in search_path {. osdla_syn_20171004_1853/src}. (VER-41)
*** Presto compilation terminated with 1 errors. ***
0
if { $analyzeStatus != 1 } {
	puts "${synMsgErr} Analyze failed!  Aborting..."
	exit 1
}
Error: Analyze failed!  Aborting..._

I checked the **osdla_syn_20171004_1853** folder. The files do exist. Can someone please help me.

nodushiv, 2017-10-06:
Has this file been generated?
>>>osdla_syn_20171004_1853/scripts/NV_NVDLA_partition_a.files.vc

If so, can you paste the content?

If not, can you attach your run log (stdout) - before design compiler was launched?

sethu018, 2017-10-06:
-y osdla_syn_20171004_1853/src
+incdir+osdla_syn_20171004_1853/src
+libext+.v
+libext+.sv
+libext+.gv
+define+DISABLE_TESTPOINTS
+define+NV_SYNTHESIS
+define+RAM_INTERFACE
NV_NVDLA_partition_a.v
-v osdla_syn_20171004_1853/src/NV_NVDLA_XXIF_libs.v

Contents of osdla_syn_20171004_1853/scripts/NV_NVDLA_partition_a.files.vc

nodushiv, 2017-10-06:
That seems about right. 

Can you attach the run log? (the stdout before synthesis actually began)?

sethu018, 2017-10-06:
[INFO]: Sourcing default flow variables from ./syn/scripts/default_config.sh ...
[INFO]: Sourcing user synthesis configuration file ./syn/templates/config.sh ...
[INFO]: Creating sandbox osdla_syn_20171004_1853 ...
[INFO]: Copying flow source code into osdla_syn_20171004_1853/scripts/ ...
[INFO]: Searching for RTL with extension: .v .sv .gv
[INFO]: Searching for INCLUDE files with extension: .vh .svh
[INFO]: Copied all RTL and include files into osdla_syn_20171004_1853/src
[INFO]: Removing any designware components from osdla_syn_20171004_1853/src
[INFO]: Generated module input dependency file osdla_syn_20171004_1853/scripts/NV_NVDLA_partition_a.files.vc
[INFO]: Generated module input dependency file osdla_syn_20171004_1853/scripts/NV_NVDLA_partition_c.files.vc
[INFO]: Generated module input dependency file osdla_syn_20171004_1853/scripts/NV_NVDLA_partition_o.files.vc
[INFO]: Generated module input dependency file osdla_syn_20171004_1853/scripts/NV_NVDLA_partition_m.files.vc
[INFO]: Generated module input dependency file osdla_syn_20171004_1853/scripts/NV_NVDLA_partition_p.files.vc
[INFO]: Running DC (non-physical/Wireload model)...
/apps/common/synopsys/syn_2013.12/bin/dc_shell -no_gui -f osdla_syn_20171004_1853/scripts/dc_run.tcl -output_log_file osdla_syn_20171004_1853/log/NV_NVDLA_partition_a_wlm.log
Information: License queuing is enabled. (DCSH-18)

I hope this is what you had meant. If not, please guide me on where to find the run log

nodushiv, 2017-10-07:
Yes, that is what I was looking for - but looking at that - It doesn't seem like there is anything wrong going on. 

Can you also provide your config file? Can you make sure that all the RTL is copied under 
>> osdla_syn_20171004_1853/src 

If not, can you double check the value of RTL_SEARCH_PATH and NVDLA_ROOT in your config file?

sethu018, 2017-10-07:
Hi the following is my config file

export TOP_NAMES="NV_NVDLA_partition_a NV_NVDLA_partition_c NV_NVDLA_partition_o NV_NVDLA_partition_m NV_NVDLA_partition_p"

export NVDLA_ROOT="/ece/home/sethu018/work/nvidia/hw-master"

# Where do I find the RTL source verilog/system verilog files?
export RTL_SEARCH_PATH=" \
	$(ls -d ${NVDLA_ROOT}/vmod/nvdla/*) \
	${NVDLA_ROOT}/vmod/rams \
	${NVDLA_ROOT}/vmod/vlibs \
"

# For verilog source files that do not match the module name.
export EXTRA_RTL=" \
	${NVDLA_ROOT}/vmod/nvdla/nocif/NV_NVDLA_XXIF_libs.v \
"

# If there are verilog header files, where do I find them?
export RTL_INCLUDE_SEARCH_PATH=" \
	${NVDLA_ROOT}/vmod/include \
"

# File extensions for source files...
export RTL_EXTENSIONS=".v .sv .gv"
export RTL_INCLUDE_EXTENSIONS=".vh .svh"

# FLoorplans and constraints
export DEF="def"
export CONS="cons"

# ===========================
# TOOL RELATED VARIABLES
# ===========================

# Design Compiler Installation - Where do I find the dc_shell executable
export DC_PATH="/apps/common/synopsys/syn_2013.12/bin"

# ===========================
# LIBRARY RELATED VARIABLES
# ===========================

export RELEASE_DIR="/project/sartori00/LibDir"
export TARGET_LIB="\
 	${RELEASE_DIR}/db/NVT_100.db \
 	${RELEASE_DIR}/db/LVT_100.db \
 	${RELEASE_DIR}/db/HVT_100.db"
export LINK_LIB="\
 	${RELEASE_DIR}/db/NVT_100.db \
 	${RELEASE_DIR}/db/LVT_100.db \
 	${RELEASE_DIR}/db/HVT_100.db"
# ==========================
# MISCELLANEOUS VARIABLES
#===========================
# Set host options in the DC session.
export DC_NUM_CORES="8"

# Apply constraints to tighten CG enable paths to model post-CTS insertion delays
export TIGHTEN_CGE="1"

# Enable Area recovery (run optimize_netlist -area)
export AREA_RECOVERY="1"

# Number of incremental recompile loops
export INCREMENTAL_RECOMPILE_COUNT="2"

# For Job management
#export COMMAND_PREFIX="qsub -q o_timing_cpu_32G -app syn -n 8 -R span[hosts=1] -oo <LOG>/<MODULE>.lsf.log -N "
export COMMAND_PREFIX=""

export CGLUT_FILE="$NVDLA_ROOT/templates/cg_latency_lut.tcl"

sethu018, 2017-10-07:
And my NVDLA_ROOT is set to /ece/home/sethu018/work/nvidia/hw-master
             RTL_SEARCH_PATH to /ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/apb2csb
I have also verified that all source files are there in **osdla_syn_20171004_1853/src** directory

nodushiv, 2017-10-07:
> RTL_SEARCH_PATH to /ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/apb2csb

That's not correct. 

You need to have all search paths included in RTL_SEARCH_PATH - This should be fine:

>export RTL_SEARCH_PATH=" 
$(ls -d ${NVDLA_ROOT}/vmod/nvdla/*) 
${NVDLA_ROOT}/vmod/rams 
${NVDLA_ROOT}/vmod/vlibs 
"

**I am unable to reproduce your error locally.** 

Looking at the message you posted originally:
>Searching for osdla_syn_20171004_1853/src/NV_NVDLA_partition_a.v
Error: Unable to open file `NV_NVDLA_partition_a.v': in search_path {. osdla_syn_20171004_1853/src}. (VER-41)

That means NV_NVDLA_partition_a.v is not present in this dir:
>osdla_syn_20171004_1853/src

Which means it was not copied over. 

Perhaps you are missing some files. 

Can you run this and let me know what you find?

>$> find /ece/home/sethu018/work/nvidia/hw-master -name '*.v' | wc -l

It should return a count of **406**

>$> find osdla_syn_20171004_1853 -name '*.v' | wc -l 

It should return a count of **372**

If you aren't seeing the same counts, then you are missing something.

sethu018, 2017-10-07:
**find /ece/home/sethu018/work/nvidia/hw-master -name '*.v' | wc -l**  given me 388

there are 372 .v files in vmod directory and 16 in verif folder.

**find osdla_syn_20171004_1853 -name '*.v' | wc -l** gives me 372.

sethu018, 2017-10-07:
And for RTL_SEARCH_PATH, the following are defined:

/ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/apb2csb
/ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/bdma
/ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/cacc
/ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/car
/ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/cbuf
/ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/cdma
/ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/cdp
/ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/cmac
/ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/csb_master
/ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/csc
/ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/glb
/ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/nocif
/ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/pdp
/ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/retiming
/ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/rubik
/ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/sdp
/ece/home/sethu018/work/nvidia/hw-master/vmod/nvdla/top 
/ece/home/sethu018/work/nvidia/hw-master/vmod/rams 	
/ece/home/sethu018/work/nvidia/hw-master/vmod/vlibs

I had earlier printed only the first line by mistake

nodushiv, 2017-10-07:
>there are 372 .v files in vmod directory 
Your count seems off. Are you sure you have the whole tree synced?

This is what I see:
`$> find hw/vmod -name '*.v' | wc -l
406`

This is in a clone I _just_ created. 

You might want to checkout afresh. 

Are you hardcoding all the paths in RTL_SEARCH_PATH?

That will work, but is unnecessary, really. This, from the template, should work just fine:
```
export RTL_SEARCH_PATH=" 
$(ls -d ${NVDLA_ROOT}/vmod/nvdla/*) 
${NVDLA_ROOT}/vmod/rams 
${NVDLA_ROOT}/vmod/vlibs 
"
```

The wildcard should take care of finding all paths in the current release structure.

sethu018, 2017-10-07:
export RTL_SEARCH_PATH=" 
$(ls -d ${NVDLA_ROOT}/vmod/nvdla/*) 
${NVDLA_ROOT}/vmod/rams 
${NVDLA_ROOT}/vmod/vlibs 
"

This is what is being called in script. I printed the lines from log file. 

I just pulled the files again. I see 406 files in vmod. I shall try running synthesis again. I shall get back if there are any errors.

sethu018, 2017-10-07:
I still do get the same error in synthesis of latest files. Kindly let me know if I need to check something else. 
Thanks

nodushiv, 2017-10-07:
Are you still seeing the same counts? 
(406 .v files in hw/vmod - and 372 .v files in the <synth_build_tag>/src directory)

Sorry - Unfortunately, I am unable to reproduce the issue locally.

sethu018, 2017-10-08:
Hi,

  I observed that the .v files in ram weren't being copied into <synth_build_tag>/src. It was because in config.sh, the RTL_SEARCH_PATH was given as:

**export RTL_SEARCH_PATH="
$(ls -d ${NVDLA_ROOT}/vmod/nvdla/*)
${NVDLA_ROOT}/vmod/rams
${NVDLA_ROOT}/vmod/vlibs
"**

in rams directory there were two other directories called model and synth. Hence I made the following change:
**${NVDLA_ROOT}/vmod/rams/***

Now both vmod and <synth_build_tag>/src have 406 files.

But still the error hasn't been resolved.

nodushiv, 2017-10-08:
The template calls for `hw/vmod/rams/synth` to be on `RTL_SEARCH_PATH`

That is intentional. 

The RAM*.v files in `hw/vmod/rams/model` are not synthesizable, and should not be picked up for synthesis. 

That is why the template only includes `hw/vmod/rams/synth`

We expect that users would provide timing models or synthesizable models for these RAM macros, as mentioned in the integrators guide:
http://nvdla.org/integration_guide.html#memories-sram

I'm not sure how I can debug further without actually seeing your setup  :)

I had a couple of folks try it out here, and they aren't seeing any issues. 

Can I have you zip the entire directory as is, and attach it here, including DC logs?

sethu018, 2017-10-08:
Hi,

  With the default template for RTL_SEARCH_PATH, there are only 316 files in <synth_build_tag>/src. The 90 files in rams directory are not included.
Probably i might need to add synth directory in the template as well?

I am unable to attach the entire zip of my setup in this comment. Shall I send it via email?

sethu018, 2017-10-08:
I am unable to send it via email as well. I can upload in google drive and share it with you.

nodushiv, 2017-10-08:
The template calls for including rams/synth, and not rams/model. 
[image]

Please share via google drive.

sethu018, 2017-10-08:
Hi,

  I have shared the files to **notifications@github.com**

Thanks

nodushiv, 2017-10-08:
I wouldn't receive it if you sent it to that email.

Please paste the link here, and I can request access. 
https://support.google.com/drive/answer/2494822?hl=en&ref_topic=7000947

sethu018, 2017-10-08:
Kindly access the files from the following link:
 
https://drive.google.com/drive/folders/0B71aFDnnWgsRT0tkNlI5UEpRSlk?usp=sharing

nodushiv, 2017-10-09:
I see from your config file:
[image]

That you have vmod/rams in the search path, not **vmod/rams/synth**

As I pointed out, rams/synth needs to be in the search path.

Once I fixed that, I was able to run synthesis fine with your setup (it is past the link stage).

sethu018, 2017-10-09:
Hi , 

   I did try the same. It still seems to encounter the same error. Could you please tell me what pathnames you had changed in config file from my setup in order to run the synthesis at your end. 

   Also, could other versions of Design Compiler create such problems?

Thanks,
Subhash

nodushiv, 2017-10-09:
I only changed RTL_SEARCH_PATH, DC_PATH and NVDLA_ROOT from your setup. 

I see that you are using DC 2013.12-SP1 - Let me try and verify that version. We've only tested out 2016.12 versions and newer.

nodushiv, 2017-10-09:
I just checked 2013.12-SP1 and a few other DC versions - I am not seeing any problems.

sethu018, 2017-10-10:
Hi,

  I made the following change in syn_launch.sh script. In the code where module.files.vc is created,

**#echo "+incdir+$BUILD_NAME/src" >> $BUILD_NAME/scripts/${module}.files.vc
echo "$BUILD_NAME/src/$module.v" >> $BUILD_NAME/scripts/${module}.files.vc**

   In this way, the file with whole path name shall be searched rather than searching in the specified directory. 
   I tried this and DC was able to read the file **/ece/home/sethu018/work/nvidia-ver-2/hw-master/osdla_syn_20171009_1204/src/NV_NVDLA_partition_a.v**

But, I am seeing the following warnings:

**set vcsOpt "{-f $RTL_DEPS}"
{-f /ece/home/sethu018/work/nvidia-ver-2/hw-master/osdla_syn_20171009_1204/scripts/NV_NVDLA_partition_a.files.vc}
puts "cwd = [pwd]"
cwd = /ece/home/sethu018/work/nvidia-ver-2/hw-master
catch {eval {analyze -format sverilog -work WORK} -vcs $vcsOpt} analyzeStatus
Running PRESTO HDLC
Compiling source file /ece/home/sethu018/work/nvidia-ver-2/hw-master/osdla_syn_20171009_1204/src/NV_NVDLA_partition_a.v
Presto compilation completed successfully.
0
if { $analyzeStatus != 1 } {
	puts "${synMsgErr} Analyze failed!  Aborting..."
	exit 1
}
# Elaborate and switch design to current block.
elaborate ${MODULE}
Running PRESTO HDLC
Presto compilation completed successfully.
Elaborated 1 design.
Current design is now 'NV_NVDLA_partition_a'.
Information: Building the design 'NV_NVDLA_reset'. (HDL-193)
Warning: Cannot find the design 'NV_NVDLA_reset' in the library 'WORK'. (LBR-1)
Information: Building the design 'NV_NVDLA_sync3d'. (HDL-193)
Warning: Cannot find the design 'NV_NVDLA_sync3d' in the library 'WORK'. (LBR-1)
Information: Building the design 'NV_NVDLA_sync3d_s'. (HDL-193)
Warning: Cannot find the design 'NV_NVDLA_sync3d_s' in the library 'WORK'. (LBR-1)
Information: Building the design 'NV_NVDLA_cacc'. (HDL-193)
Warning: Cannot find the design 'NV_NVDLA_cacc' in the library 'WORK'. (LBR-1)
Information: Building the design 'NV_NVDLA_RT_cmac_b2cacc'. (HDL-193)
Warning: Cannot find the design 'NV_NVDLA_RT_cmac_b2cacc' in the library 'WORK'. (LBR-1)
Warning: Design 'NV_NVDLA_partition_a' has '5' unresolved references. For more detailed information, use the "link" command. (UID-341)**

Is there a way I can resolve that?

sethu018, 2017-10-11:
Hi,

In the last message I mentioned that i had commented the +incdir line in syn_launch.sh
I assumed the modules were missing because of not including the directory where the source files are. 

Hence, I added that back again (uncommented it). SO the only change I have made to syn_launch.sh is dump the whole pathname of the file in .vc file.

**echo "$BUILD_NAME/src/$module.v" >> $BUILD_NAME/scripts/${module}.files.vc**

But still, the same error exists as the previous message.

Thanks,
Subhash

sethu018, 2017-10-12:
Hi,
  
    I was able to resolve the previous issue. I had to run on a different machine to resolve that. 

But I am now facing a new issue where there is an error in link stage:

**Information: Building the design 'DW_lsd' instantiated from design 'NV_NVDLA_CACC_CALC_fp_48b' with
    	the parameters "a_width=36". (HDL-193)
Warning: Cannot find the design 'DW_lsd' in the library 'WORK'. (LBR-1)
Information: Building the design 'DW_lsd' instantiated from design 'NV_NVDLA_CACC_CALC_fp_48b' with
    	the parameters "a_width=44". (HDL-193)
Warning: Cannot find the design 'DW_lsd' in the library 'WORK'. (LBR-1)
Information: Building the design 'RAMPDP_32X288_GL_M1_D2'. (HDL-193)
Warning: Cannot find the design 'RAMPDP_32X288_GL_M1_D2' in the library 'WORK'. (LBR-1)
Information: Building the design 'RAMPDP_32X192_GL_M1_D2'. (HDL-193)
Warning: Cannot find the design 'RAMPDP_32X192_GL_M1_D2' in the library 'WORK'. (LBR-1)
Information: Building the design 'RAMPDP_32X256_GL_M1_D2'. (HDL-193)
Warning: Cannot find the design 'RAMPDP_32X256_GL_M1_D2' in the library 'WORK'. (LBR-1)
Information: Building the design 'RAMPDP_32X224_GL_M1_D2'. (HDL-193)
Warning: Cannot find the design 'RAMPDP_32X224_GL_M1_D2' in the library 'WORK'. (LBR-1)
Warning: Unable to resolve reference 'DW_lsd' in 'NV_NVDLA_CACC_CALC_fp_48b'. (LINK-5)
Warning: Unable to resolve reference 'RAMPDP_32X288_GL_M1_D2' in 'nv_ram_rws_32x768_logic_0'. (LINK-5)
Warning: Unable to resolve reference 'RAMPDP_32X192_GL_M1_D2' in 'nv_ram_rws_32x768_logic_0'. (LINK-5)
Warning: Unable to resolve reference 'RAMPDP_32X288_GL_M1_D2' in 'nv_ram_rws_32x544_logic_0'. (LINK-5)
Warning: Unable to resolve reference 'RAMPDP_32X256_GL_M1_D2' in 'nv_ram_rws_32x544_logic_0'. (LINK-5)
Warning: Unable to resolve reference 'RAMPDP_32X288_GL_M1_D2' in 'nv_ram_rws_32x512_logic_0'. (LINK-5)
Warning: Unable to resolve reference 'RAMPDP_32X224_GL_M1_D2' in 'nv_ram_rws_32x512_logic_0'. (LINK-5)
Error: Failed to link. Aborting...**

Could you please help me with this?

Thanks,
Subhash

nodushiv, 2017-10-12:
Could you provide details about the machine which was running into failures? And the machine that was passing?
* OS version
* CPU arch and model
* which shell, version of the shell

As I mentioned before, we do not release the synth models for the RAMs. You will need to provide the synthesizable models, or timing models for them.

As for the DW_lsd - Do you have DesignWare licenses? If so, can you check if the design ware library is being loaded in the DC session? Please attach the log file.

sethu018, 2017-10-12:
Hi,

The specification of machine where there was read error:
OS: Red Hat Enterprise Linux Workstation release 6.9
CPU arch: x86_64
CPU model: Intel(R) Xeon(R) CPU E5-2640 v2 @ 2.00GHz
Shell: tcsh

The specification of machine where read was successful:
OS: Red Hat Enterprise Linux Workstation release 7.4 
CPU arch: x86_64
CPU model: Intel(R) Xeon(R) CPU E5-2620 v4 @ 2.10GHz
Shell: tcsh

Attaching log file.

[dc_log.txt](https://github.com/nvdla/hw/files/1381012/dc_log.txt)

Could you please let me know where I can find the timing model for RAM?

nodushiv, 2017-10-12:
If you do not have access to DesignWare - Please refer this section from the integrator's manual:
http://nvdla.org/integration_guide.html#designware-components

You will need to edit the RTL to enable a verilog define called
`DESIGNWARE_NOEXIST `

As for the RAM* modules - you will need to provide a synthesis model or timing models.

nodushiv, 2017-10-12:
Unless there are other issues, I am going to close this one. Please feel free to reopen or open a new issue.

closed 2017-10-12 by nodushiv

sethu018, 2017-10-12:
I have set the environment variable DESIGNWARE_NOEXIST to 1. Do I have to define it in any other file?

jedimaster2001, 2021-02-18:
Hi, I ran into this "Cannot find the design...in  the library 'WORK'" due to the way I use 'parameter' in the module instantiation.

In dc_shell run log:
Information: Building the design 'XXX'. (HDL-193)
Warning: Cannot find the design 'XXX' in the library 'WORK'. (LBR-1)
Information: Building the design 'XXX' instantiated from design '...' with
	the parameters "17,32". (HDL-193)
Warning: Cannot find the design 'xxx' in the library 'WORK'. (LBR-1)

The tool doesn't seem to understand the one in comment, which is valid for Verilog-2001
module_name  #(17,32) 
//#( .ADDR(17),
//   .DATA(32)
//) instance_name (....)

Not sure why it would not take Verilog-2001 syntax.

Allenjin123, 2023-04-20:
Hi! I'm facing exact the same problem about parameter redefinition. Do you have any solution afterwards?