nvdla/hw #24: Q:the input.txn's register map is different from http://nvdla.org/hwarch.html

huangleiesc, 2017-10-09:
when i run the nvdla testcase,i can't understand the full meaning in input.txn. I mean i can't find the register   
 in "Address space layout". for example,"write reg 0xfff1401 0x0", i can’t find the description for 0x1401，except the comment.

huangleiesc, 2017-10-09:
i find the 0x1401 is map to 0x9000 in CSC, is it right? how can i get more information about register? i want to know the detail about the case "conv_8x8_fc_int16".

huangleiesc, 2017-10-09:
i find the 0x1401 is map to 0x8004 in CMDA, is it right? how can i get more information about register? i want to know the detail about the case "conv_8x8_fc_int16".

zdraw, 2017-10-16:
The inconsistency is that the hardware expects you to have subtracted the global offset and then shifted the result by 2 to get the word address. 0x1401 = (0x8004 - 0x3000) >> 2.

input.txn is a list of register and memory accesses that will be done by the test before finishing the simulation in what we call "trace" format. The script in synth_tb/sim_scripts/inp_txn_to_hexdump.pl reads input.txn and outputs input.txn.raw containing encodings for the csb master sequencer to read and execute from. Documentation for the trace format is here https://github.com/nvdla/doc/blob/master/doc/integration_guide.rst#testbench--traces and I can update it with my more detailed answer here.
Documentation for the CSB protocal is here: https://github.com/nvdla/doc/blob/master/doc/hwarch.rst#configuration-space-bus

"write_reg 0xffff1401 0x0" is decoded as 
1. "write_reg" cmd
2. "0xffff1401" address
  * The lower 16 bits are the word address of the register offset by the NVDLA_GLB.S_NVDLA_HW_VERSION_0 reg offset. 
  ** So you take 0x8004, subtract 0x3000 to get 0x5004, shift right by 2 for the word address, gives 0x1401.
  * Since this txn is played on the CSB interface, CSB protocol has extra command bits
  * The upper 16 bits are extra bits that decode into 
  ** Only bit of this that actually matters is bit 17 of bits 31 to 16 which corresponds to nposted
  ** Bit 16 is a rd/wr bit set by the inp_txn_to_hexdump script
  ** Bits 31 to 18 are not used 
  ** I set these 16 bits to 1 since I wanted nposted to be 1 and the others don't matter and needed to double check that.
  ** See below for code snippet from csb_master_seq.v that uses this
  
3. "0x0" data

csb_master_seq.v
MSEQ_REG_WR: begin
            mseq_pending_req <= 1;
            mseq2mcsb_pd[62:55] <= curr_cmd[`MSEQ_PD_BITS];                    //further defined below
            mseq2mcsb_pd[54] <= 1; //Writes set this bit to 1                          //1bit rd/wr indicator, set by inp_txn_to_hexdump.pl automatically
            mseq2mcsb_pd[21:0] <= {6'b0,curr_cmd[`MSEQ_ADDR_PD_BITS]}; //16bit addr extended to 22
            mseq2mcsb_pd[53:22] <= curr_cmd[`MSEQ_DATA_PD_BITS];          //32 bit data

csb_master.v
assign mcsb2scsb_pd_level = mcsb2scsb_pd[62:61]; //2 bit Priority **Not used in this version of NVDLA
assign mcsb2scsb_pd_wrbe = mcsb2scsb_pd[60:57]; //4 bit write byte enable for the 4 bytes of data
assign mcsb2scsb_pd_srcpriv = mcsb2scsb_pd[56:56]; // 1 bit privilege **Not used in this version of NVDLA
assign mcsb2scsb_pd_nposted = mcsb2scsb_pd[55:55]; // 1bit. Set to 1 if you want to watch for a write_complete or error returned from the design.
assign mcsb2scsb_pd_write = mcsb2scsb_pd[54:54];  // 1bit. Set to 1 for writes and 0 for reads.
assign mcsb2scsb_pd_wdat = mcsb2scsb_pd[53:22];
assign mcsb2scsb_pd_addr = mcsb2scsb_pd[21:0];

zdraw, 2017-10-16:
We're working on documentation with more details about the registers. Stay tuned.

closed 2017-10-16 by zdraw