nvdla/hw #34: How can i use the input_feature_map.dat

wumh, 2017-10-13:
In ~/hw_mast/verif/sim/

make build DUMP=1 DUMPER=VERDI
make run DUMP=1 DUMPER=VERDI TESTDIR=../traces/first_release
make verdi DUMP=1 DUMPER=VERDI TESTDIR=../traces/first_release

There is a file name“input_feature_map.dat” in the directory "hw_mast/verif/sim/conv_8x8_fc_int16"that seems to be used as a  memory input in the simulation phase.
The question is, how should I use it?

huangleiesc, 2017-10-16:
i think you can do "make run DUMP=1 DUMPER=VERDI TESTDIR=../traces/first_release/conv_8x8_fc_int16"

zdraw, 2017-10-16:
huangleiesc is correct in the usage, the TESTDIR needs to point to a directory containing input.txn, which is a file containing memory files to load at specific locations, as well as registers to read/write.

In case your question was more about what is input_feature_map.dat used for.. it's a file containing arbitrary data that conv_8x8_fc_int16 uses with input_weight.dat, also arbitrary for this testcase, to perform a convolution. The resulting output is in output_feature_map.dat. The test will dump out the memory region specified in input.txn and compare against output_feature_map.dat.

closed 2017-10-16 by zdraw