nvdla/hw #35: Q:how can i use softmax with nvdla

huangleiesc, 2017-10-16:
i can‘t see the div operation in nvdla.

ruotongyu, 2017-10-16:
Hi, sorry for interrupting you. I was trying to build and run the single test. But the compile failed because missing vcs tool. Do you have any idea where I can get free vcs tool? Feel free to contact me from email ruotoy@uw.edu

jwise, 2017-10-16:
@huangleiesc Currently, I believe that we expect that softmax will be emulated on CPU.  @nvdsmith can you confirm?

@ruotongyu Please open a new bug for unrelated issues.  (In this case, we expect that users will have to acquire licenses to tooling on their own.)

nvdsmith, 2017-10-16:
Correct that softmax would be expected to be run on the host processor of the SoC which contains the NVDLA.

closed 2017-10-16 by jwise