nvdla/hw #37: Is there a sim script for Modelsim?

000xjy000, 2017-10-18:
Hi, 
I tried to run a simulation in Modelsim refer to 'sim/Makefile', but encountered a lot of PSL assertion errors and some mismatches between the expected data and mine. So I hope there is a sim script for Modelsim and I can do the test more easily.
Thanks

jwise, 2017-10-18:
We don't currently officially support ModelSim (I don't have a license for it here), but we have been making changes for tool support on a best-effort basis.

Mismatches are going to be hard to debug, but if you see other issues that you can pinpoint, we'd be happy to make changes and add workarounds as needed.  And, if you have sim scripts for Modelsim once you get it working, we'd be happy to add those to the repository.

Thanks!

xalogic-linus, 2017-11-09:
I managed to get simulations to work on Vivado Xsim (Single run only, not regression). It would require some ifdefs on testbench files particularly in the usage of $value$args. VCS implemented it differently from VIVADO. 

@jwise would you be keen to support this?

If you are supportive, I will work on a pull request soon.

jwise, 2017-11-09:
@xalogic-linus This would be great!  We'd be happy to take a pull request for that.

jevogel, 2018-01-24:
FYI, the [Intel FPGA Starter Edition of ModelSim](http://dl.altera.com/?product=modelsim_ae#tabs-2) is free to use with no license file required.

nvhook, 2018-07-03:
Closing issue

closed 2018-07-03 by nvhook