nvdla/hw #38: Q:one time one frame

huangleiesc, 2017-10-20:
in edge device,the ddr is limit and the process latency is sensitive.
if we do batch Calculation
1,the cost of ddr will increase with the number of batch.And in the CNN,the number of datas are also large.and the hidden layer data are more large ,due to the increase of channel.
2,when we collect  16 frames from sensor in the 60fps,the delay is near 300ms. Maybe in some apply,the delay is too long.
so i think ,in the edge device ,the common batch is 1 or 2. in this case ,i see the nvdla's Network MAC Utilization is low.Are you consider to optimization the performance in this case,that batch is low

rill-zhen, 2017-10-20:
channel extension feature maybe useful.

nvdsmith, 2017-10-20:
NVDLA doesn't use batching during convolution to fully utilize MAC's.   Is your concern related to fully connected layers?  By their nature, fully connected layers will require higher memory bandwidth for weights, and will benefit from batching.

huangleiesc, 2017-10-24:
thank you. NVDLA doesn't use batching during convolution to fully utilize MAC's,so there is no problem in my question 1.

closed 2017-10-24 by huangleiesc