nvdla/hw #40: is there a data path from PDP to CDP?

rill-zhen, 2017-10-24:
is there a data path from PDP to CDP?
we can see that there is a data path from PDP to CDP （according to “i.e., the convolution core can pass data to the Single Data Point Processor, which can pass data to the Planar Data Processor, and in turn to the Cross-channel Data Processor” & the architecture diagram, at: http://nvdla.org/primer.html#hardware-architecture) , actually, there is not data path between PDP and CDP according to the RTL source released on github.
there maybe a mistake in the Primer document.
Please double check it.

zhangpengsjtu, 2017-11-06:
There is no such a path from PDP to CDP, we will fix the Doc.

nvhook, 2018-07-03:
Closing issue as resolved.

closed 2018-07-03 by nvhook