nvdla/hw #41: low MAC utilization for dw_conv in MobileNet

rill-zhen, 2017-10-24:
i know the mechanism of the CMAC according to reading the RTL source, and realized that the MAC utilization will be very low in such a scenario(dw convolution). is there a solution for this?
does this requires significant modifications? any ideas?

nvdsmith, 2017-11-30:
Right, for DW convolution the MAC utilization would be low.  The 1x1 convolutions in mobilenet would be handled efficiently.

Using winograd would improve utilization for the DW layers.  For example, the small configuration we're planning will have 16 multipliers per MAC-cell, and 4 MAC-cells.  Winograd on a 3x3 dw convolution would utilize 16 multipliers from one mac cell.  So overall 25% utilization.

jwise, 2018-02-06:
Since there hasn't been any activity recently, I'm going to close this bug; if you still have issue, please feel free to reopen it.

closed 2018-02-06 by jwise