nvdla/hw #42: the format of feature and coefficient in DRAM/CBUF

rill-zhen, 2017-10-26:
as we known, the format of freature and weight is very important to understand the architecture of NVDLA. Could you please give a brief description of the format，thanks in advance.

zdraw, 2017-10-30:
The .dat files as parsed for lines with 32 groups of hex bytes. First byte on the line of 32 maps to offset 0 (the base address is specified in input.txn) and the last byte on the line maps to offset 31. The first byte of the second line maps to offset 32 and the last byte on the line maps to offset 63, etc.
Lines without "0x" at the beginnning are ignored so some of our files have comments in plain text.
When reading a .dat file, the format is
0x<byte0> 0x<byte1> ... 0x<byte30> 0x<byte31>
0x<byte32> 0x<byte33> ... 0x<byte62> 0x<byte63>

verif/synth_tb/sim_scripts/input_txn_to_hexdump.pl does the conversion of .dat to .raw2 for simulation. The testbench takes in .raw2 files using readmemh. readmemh takes values 4 bytes at a time and the right-most byte going to address 0. 
When reading a .raw2 files, the format is
<byte7><byte6>...<byte1><byte0>
<byte15><byte14>...<byte9><byte8>

rill-zhen, 2017-10-31:
the 'format' here, i am refer in particular to the structure of the feature-maps and the weights in the Memory and CBUF.for example ,for feature-map(Width x Height x M_channel) , weight(Row x Column x M_channel x N_channel), 
is 0xbyte0 the weight[w0][h0][m0][n0]? 
is 0xbyte2 the weight[w1][h0][m0][n0] or weight[w1][h0][m1][n0]?
where is the weight[w_i][h_j][m_k][n_l] stored in the Memory and CBUF?

jwise, 2018-01-08:
We've written some documentation about this: http://nvdla.org/hw/format.html

closed 2018-01-08 by jwise