nvdla/hw #43: Checktest : FAILED : conv_8x8_fc_int16, Found mismatches

tutikudsi, 2017-10-27:
While running the conv_8x8_fc_int16 test case I am getting the result as failure. It gives the following message when I check it with the command "make check TESTDIR=../traces/traceplayer/conv_8x8_fc_int16"

" checktest : FAILED : conv_8x8_fc_int16                        : all transactions completed with no errors but dump_mem mismatched: Found mismatches between conv_8x8_fc_int16/0.chiplib_dump.raw2 and conv_8x8_fc_int16/0.chiplib_replay.raw2 "

The contents of the 0.chiplib_dump.raw2 are all zeros. Where as the contents of the file 0.chiplib_replay.raw2 are the contents of the file output_feature_map.dat, as was mentioned in issue https://github.com/nvdla/hw/issues/34

While going through the log I found a warning saying:
"Warning-[STASKW_RMTMDWIFAL] Too many data words in file
/nvdla/hw-master/verif/synth_tb/csb_master_seq.v, 419
  Too many data words in file 0.raw2 at line 1026 while executing $readmem.
  Please ensure that the file has proper entries."

When I checked the 0.raw2 file it had following contents at the respective lines
1025: 1745c349
1026: 056d9f98
1027: faf4fc31

Can someone please help me fix this issue?

xranger, 2017-10-29:
same issue as you, have you solved this?

tutikudsi, 2017-10-29:
Not yet

zdraw, 2017-10-30:
I can't recreate your "make check" error but I do have the Warning with an overall PASS, my dump and replay match so the Warning is not the issue (but something I'll look into).
test.log:Warning-[STASKW_RMTMDWIFAL] Too many data words in file
test.log-/home/scratch.stephenh_t194/nvdla_git_github/hw/verif/synth_tb/csb_master_seq.v, 419
test.log-  Too many data words in file 0.raw2 at line 1026 while executing $readmem.

What's odd is that your 0.chiplib_dump.raw2 has 0's. This is created by verif/synth_tb/sim_scripts/inp_txn_to_hexdump.pl and should be the contents of output_feature_map.dat. Perhaps you mixed up the files in your comment?

**Can you check two things, the output of inp_txn_to_hexdump.pl and the end message of test.log?**
sim/conv_8x8_fc_int16> ../../synth_tb/sim_scripts/inp_txn_to_hexdump.pl ../../traces/traceplayer/conv_8x8_fc_int16 
default 32 memory width
accessing ../../traces/traceplayer/conv_8x8_fc_int16/input.txn
File 1 0x80000000 0x1000 0
File 2 0x80100000 0x10000 0
File 2 0x80400000 0x20 0

The end of the tests should look like this, does it match your output?
352110.00ns SMEM: Slave wrote address 0x0080400000 data 0xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx7fff7fff7fff00007fff00007fff7fff00007fff7fff00007fff000000000000
AXI_MEM_XACTION: WriteResp: CHANNEL=0 TIME=352230.00ns LATENCY=1120.00ns BID=0x01
352330.00ns MSEQ: Write (command 223) completed (addr: 0xffff0003, data: 0xffffffff.
352410.00ns MSEQ: read_cmd address 0xffff0003 with data 0x00000001 and mask 0x00000001 (command 224)
353210.00ns MSEQ: Read (command 224) matched (addr: 0xffff0003, received data: 0x00000001, expected data: 0x00000001.
353290.00ns MSEQ: write_cmd address 0xffff0003 with data 0xffffffff (command 225)
354090.00ns MSEQ: Write (command 225) completed (addr: 0xffff0003, data: 0xffffffff.
354170.00ns MSEQ: Backdoor mem_dump of file            0.chiplib_replay.raw2 at address 0x80400000 for length 0x00000020.
$finish called from file "/home/scratch.stephenh_t194/nvdla_git_github/hw/verif/synth_tb/csb_master_seq.v", line 274.
$finish at simulation time     354710.00ns
           V C S   S i m u l a t i o n   R e p o r t
Time: 354710000 ps

tutikudsi, 2017-10-31:
Hi @zdraw , thanks a lot for your response.

I was able to resolve this issue. The problem was related to the variable DESIGNWARE_NOEXIST in the Makefile. It was set to 0 initially. While going to the documentation again from http://nvdla.org/integration_guide.html#designware-components , what I could understand was to set it to 1. 

Hi @xranger, you can try the same. I think it should work. Will wait for your response and then close this issue.

jmge, 2017-10-31:
@tutikudsi @zdraw , thanks a lot for your input. However, it does not work to me. Whatever I set DESIGNWARE_NOEXIST, the result are the same.
below are some log from my run. 
Beignning of the test:
`[colin@cc_server1 sim]$ make run DUMP=1 DUMPER=VERDI TESTDIR=../traces/traceplayer/sanity3
/bin/mkdir -p /home/colin/work/hw-master/verif/sim/sanity3 ; /bin/cp -Trf /home/colin/work/hw-master/verif/sim/../traces/traceplayer/sanity3 /home/colin/work/hw-master/verif/sim/sanity3 ; cd /home/colin/work/hw-master/verif/sim/sanity3 ; /bin/ln -fs /home/colin/work/hw-master/verif/sim/./simv ; /bin/ln -fs /home/colin/work/hw-master/verif/sim/./simv.daidir ; \
	/bin/rm -f input.txn.raw ; \
	/bin/rm -f *.raw2 ; \
	/bin/rm -f test.log ; \
	/home/colin/work/hw-master/verif/synth_tb/sim_scripts/slave_mem.cfg.pl -outFile ./slave_mem.cfg  +continue_on_fail ; \
	/home/colin/work/hw-master/verif/synth_tb/sim_scripts/inp_txn_to_hexdump.pl  /home/colin/work/hw-master/verif/sim/sanity3 ; \
	 ./simv -l test.log  +read_reg_poll_retries=10 +continue_on_fail +continue_on_fail  +dump_fsdb +fsdb+dump_log=off +fsdbfile+/home/colin/work/hw-master/verif/sim/sanity3/debussy.fsdb +input_file=/home/colin/work/hw-master/verif/sim/sanity3/input.txn +input_dir=/home/colin/work/hw-master/verif/sim/sanity3 ; \
	/home/colin/work/hw-master/verif/synth_tb/sim_scripts/checktest_synthtb.pl ./test.log; \
	/bin/echo " " ; \
	/bin/echo "NVINFO : ===============================================================" ; \
	/bin/echo "NVINFO : To summarize the results again, run : make check TESTDIR=../traces/traceplayer/sanity3" ; \
	/bin/echo "NVINFO : ===============================================================" ; \
	/bin/echo " "
Running /home/colin/work/hw-master/verif/synth_tb/sim_scripts/slave_mem.cfg.pl
default 32 memory width
accessing /home/colin/work/hw-master/verif/sim/sanity3/input.txn
File 1 0x80000000 0x00001000 0
File 2 0x80100000 0x00010000 0
File 2 0x80400000 0x20 0
Notice: timing checks disabled with +notimingcheck at compile-time
Chronologic VCS simulator copyright 1991-2014
Contains Synopsys proprietary information.
Compiler version I-2014.03_Full64; Runtime version I-2014.03_Full64;  Oct 31 15:42 2017
*Verdi3* Loading libsscore_vcs201403.so
*Verdi3* : Disable novas_dump.log.
*Verdi3* : Enable Parallel Dumping.
VCD+ Writer I-2014.03_Full64 Copyright (c) 1991-2014 by Synopsys Inc.
(1) NVDLA_IP_INFO : Turning on fsdb dump to file debussy.fsdb
FSDB Dumper for VCS, Release Verdi3_I-2014.03, Linux x86_64/64bit, 02/10/2014
(C) 1996 - 2014 by Synopsys, Inc.
`
Here is the end of test is like this:
`xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
AXI_MEM_XACTION: ReadResp: CHANNEL=0 START_TIME=0 END_TIME=306490.00ns RID=0x09 RDATA=0xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx LATENCY=7000.00ns LEN=x
306530.00ns SMEM: Slave 0 read address 0x008010ffc0 (mem address 0x0000000000043ff0) data 0xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
AXI_MEM_XACTION: ReadResp: CHANNEL=0 START_TIME=0 END_TIME=306570.00ns RID=0x09 RDATA=0xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx LATENCY=7080.00ns LEN=x
AXI_MEM_XACTION: ReadResp: CHANNEL=0 START_TIME=0 END_TIME=306650.00ns RID=0x09 RDATA=0xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx LATENCY=7160.00ns LEN=x
307690.00ns MSEQ: write_cmd address 0xffff0003 with data 0xffffffff (command 221)
308490.00ns MSEQ: Write (command 221) completed (addr: 0xffff0003, data: 0xffffffff.
350370.00ns MSEQ: write_cmd address 0xffff0003 with data 0xffffffff (command 223)
351170.00ns MSEQ: Write (command 223) completed (addr: 0xffff0003, data: 0xffffffff.
351330.00ns SMEM: Received slave 0 write addr 0x0080400000 mask 0x00000000ffffffff data 0xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx0000000000000000000000000000000000000000000000000000000000000000
352170.00ns SMEM: Slave wrote address 0x0080400000 data 0xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx0000000000000000000000000000000000000000000000000000000000000000
AXI_MEM_XACTION: WriteResp: CHANNEL=0 TIME=352290.00ns LATENCY=1120.00ns BID=0x01
352930.00ns MSEQ: write_cmd address 0xffff0003 with data 0xffffffff (command 225)
353730.00ns MSEQ: Write (command 225) completed (addr: 0xffff0003, data: 0xffffffff.
353810.00ns MSEQ: Backdoor mem_dump of file            0.chiplib_replay.raw2 at address 0x80400000 for length 0x00000020.
$finish called from file "/home/colin/work/hw-master/verif/synth_tb/csb_master_seq.v", line 274.
$finish at simulation time     354350.00ns
           V C S   S i m u l a t i o n   R e p o r t 
Time: 354350000 ps
CPU Time:     31.600 seconds;       Data structure size: 442.7Mb
Tue Oct 31 15:42:52 2017
checktest : FAILED : .                                        : all transactions completed with no errors but dump_mem mismatched: Found mismatches between ./0.chiplib_dump.raw2 and ./0.chiplib_replay.raw2. 
 
NVINFO : ===============================================================
NVINFO : To summarize the results again, run : make check TESTDIR=../traces/traceplayer/sanity3
NVINFO : ===============================================================
`
The content of 0.chiplib_replay.raw2 is "xxxxxxxx ..."
I tried to analysis the dumped waveform, and found that there is problem in memory accessing, i.e. nvdla cannot read the data (all the read out data are XXX) from slave memory(where the TB preload the data to it initially). 
I am suspecting the issue might be caused by VCS, maybe incorrect tool version? May I know which VCS version are you using?

By the way, actually, all the tests related to memory data access are all failed in my side. :(

BR,
xranger

zdraw, 2017-10-31:
The tool version we tested with is in verif/sim/Makefile. 
# TOOL LOCATIONS
  export VCS_HOME   := /home/tools/vcs/mx-2015.09-SP2-9-T0426
  export VERDI_HOME := /home/tools/debussy/verdi3_2015.09-SP2-11

I have a version of 2014.03 that failed with your issue of x's in 0.chiplib_replay.raw2. You'll need to use a newer version of vcs. I tested it on mx-2015.09-SP2 and that was the earliest that worked.

closed 2017-10-31 by zdraw

jmge, 2017-11-01:
@zdraw , thanks for the confirmation. Maybe a suggestion that to add the tool version requirement in the further release to avoid extra debug effort and improper issue tickets. Thanks again.