nvdla/hw #46: Missing NV_HWACC_common_tick_defines.vh

xalogic-linus, 2017-11-03:
Hi,

when I look into ./hw/outdir/nv_small/vmod/include/hw/outdir/nv_small/vmod/include/NV_HWACC_NVDLA_tick_defines.vh

I see the following line

`include "NV_HWACC_common_tick_defines.vh"
But I can't seem to find NV_HWACC_common_tick_defines.vh

Is the NV_HWACC_NVDLA_tick_defines.vh even needed?

Linus

jwise, 2017-11-04:
Good question.  Let me look into it.

zhangpengsjtu, 2017-11-04:
NV_HWACC_NVDLA_tick_defines.vh is useless for now, which is for some other usage (some internal FPGA platform )

We should be able to remove this in next release, or we will fix it when we start to setup and release FPGA platform

jwise, 2017-11-04:
moving assignee to @zhangpengsjtu