nvdla/hw #48: Commit 11bc4c5 breaks vmod/nvdla/top/NV_NVDLA_partition_o.v with syntax errors

josahe, 2017-11-08:
The file NV_NVDLA_partition_o.v has been mistakenly committed using (for example) #ifdef rather than `ifdef.

Don't you run design checks before committing RTL updates?

jwise, 2017-11-08:
Hi @josahe -- you're not the first one to be confused by this, to be sure!  We added a layer that runs through the C preprocessor first.  @xalogic-linus noted this in #45, too.  Make sure that you build with `bin/tmake` -- don't use the `.v` files as is.  This does seem to be somewhat confusing, though, so we'll look into renaming files that need preprocessing to `.vx`.

We do, in fact, have a continuous integration flow; internally, we're using Jenkins for this, but we're working in the background on figuring out how to have an external CI flow that doesn't need a VCS license.  Stay tuned on that one.

josahe, 2017-11-09:
Sorry for not seeing the previously raised issue.

I actually pulled the latest commits without revisiting the documentation so failed to notice the new build flow (everything had been previously working with my own flows).

I've run the flow and I'm back up and running. Thanks!

closed 2017-11-09 by josahe