nvdla/hw #50: why the code update 11.02 is compatible with Makefile?

CrazyBingo, 2017-11-10:
Hello:
    When download the hw-master announced in 10.19, after modify the software environment variable and Designware libs, I can run the script and successfully pass the sanitys.
    While I download the hw-master update in 11.02(has modify the software environment variable and Designware libs as well), it goes wrong while I run the case. The reason why is that there is no “../../outdir/nv_Large/vmod/..”ptah, even if I  build the path "ourdir/nv_large", and put the vmod to this path, it failed again, while there is no "../vmod/nvdla/bdma/.v"， actually there is no code about it.
    So the code is incompatible with the Makefile？ or I have done sth. wrong?

zdraw, 2017-11-10:
I assume you're talking about the simulation Makefile, please correct me if I'm wrong.
The code base has started to introduce ways to configure the RTL. Part of this includes a preprocess step before the simulation Makefile can be called. 
The steps are run "make" at hw/. This sets up paths and projects (nv_large or nv_small) You may need to provide input if your tools paths are different than ours. Since you had the original nv_large code, I suggest only trying to build nv_large)
You can double check or modify the paths and projects by opening tree.make after you're done.
You should also set PERL_ROOT and put this in your PATH variable. I have to do the following:
setenv PERL_ROOT /home/utils/perl-5.8.8
setenv PATH $PERL_ROOT/bin:$PATH
Then run "tools/bin/tmake" to run the preprocessor on the design. The outputs are placed in outdir/nvlarge

At this point you should be able to run the simulation Makefile as before. I suggest doing "make clean" in hw/verif/sim to clean out the old build objects before doing "make build" to create the simv.

zdraw, 2017-11-13:
Closing. Please reopen if you're still seeing that error.

closed 2017-11-13 by zdraw

CrazyBingo, 2017-11-29:
I'm sorry to be a little busy on a business trip, Return this topic back now.
Today I have down the code of 20171121, and done as you said:
1: hw path make, input c++/perl/gcc+java path and nv_largd
2: set perl variable in .cshrc
3: then run hw/tools/bin/tmake, but there is sth. wrong tha can not load YAML.pm and interrupted.

CrazyBingo, 2017-11-29:
@zdraw