nvdla/hw #52: ERROR: defgen: file not found: .. /outdir/nv_large/spec/defs/project.def

pigHasHub, 2017-11-12:
Hi,
I'm new to nvdla. I just tried to clone the repo and follow the integration guide http://nvdla.org/integration_guide.html#env-setup for initial setup, but it failed at './tools/bin/tmake -build vmod'. I'm working on mac os. Could you please let me know what would be missing here? Thanks.

```
==============================================
files are generated under /Users/pighashub/myWork/nvdla/outdir/nv_large/spec/manual
==============================================
/usr/local/bin/g++-5 -E -undef -nostdinc -P -C nv_large.spec -o /Users/pighashub/myWork/nvdla/outdir/nv_large/spec/defs/project.def
g++-5: warning: nv_large.spec: linker input file unused because linking not done
../../tools/bin/defgen -i /Users/pighashub/myWork/nvdla/outdir/nv_large/spec/defs/project.def -o /Users/pighashub/myWork/nvdla/outdir/nv_large/spec/defs/project.h -b c
ERROR: defgen: file not found: /Users/pighashub/myWork/nvdla/outdir/nv_large/spec/defs/project.def
 at ../../tools/bin/defgen line 112.
        main::Error('file not found: /Users/pighashub/myWork/nvdla/outdir/nv_large/spe...') called at ../../tools/bin/defgen line 78
        main::gen_define('/Users/pighashub/myWork/nvdla/outdir/nv_large/spec/defs/project.def', '/Users/pighashub/myWork/nvdla/outdir/nv_large/spec/defs/project.h', 'c') called at ../../tools/bin/defgen line 69
make: *** [project.h] Error 2
../../tools/make/vmod_common.make:16: *** ../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../tools/make/vmod_common.make:16: *** ../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
../../../tools/make/vmod_common.make:16: *** ../../../outdir/nv_large/spec/defs/project.h does not exists, please make hw/spec/defs first, exiting....  Stop.
logfile: outdir/build.log
==============================================
Filehandle GEN1 opened only for input at /Library/Perl/5.18/IO/Tee.pm line 132.
==================BUILD PASS==================
Filehandle GEN1 opened only for input at /Library/Perl/5.18/IO/Tee.pm line 132.
==============================================
Filehandle GEN1 opened only for input at /Library/Perl/5.18/IO/Tee.pm line 132.
```

HKLee2040, 2017-11-13:
You may need to check whether the following required tools exist in your system or not.
Java - jdk1.7
Perl - perl-5.8.8
CPP - gcc-4.0

jarodw0723, 2017-11-13:
>> g++-5: warning: nv_large.spec: linker input file unused because linking not done
You probably shouldn't use g++, but should use cpp, which is for pre-processing. Change your tree.make 'CPP := /usr/local/bin/cpp'

pigHasHub, 2017-11-14:
Thanks @jarodw0723, @HKLee2040!
Turns out I should use 'CPP := /usr/local/bin/c++-5' in my env for pre-processing.

Just FYI, I also need to replace below line as 'chmod' does not have '--reference' option in mac os.

```
./tools/bin/eperl:
   &System ("chmod --reference=$InFile $TmpFile"); 
   ==>
   &System ("chmod `stat -f %A ${InFile}` ${TmpFile}");
```

closed 2017-11-14 by pigHasHub