nvdla/hw #53: conv_8x8_fc_int16, mem mismatched

280185386, 2017-11-15:
checktest : FAILED : .                                        : all transactions completed with no errors but dump_mem mismatched: Found mismatches between ./0.chiplib_dump.raw2 and ./0.chiplib_replay.raw2. 
 
NVINFO : ===============================================================
NVINFO : To summarize the results again, run : make check TESTDIR=../traces/traceplayer/conv_8x8_fc_int16
NVINFO : ===============================================================

jwise, 2017-11-15:
What version of vcs are you using?  We've seen some issues of this nature with older versions of vcs.

What's in the chiplib_dump/chiplib_replay file?  Is one of them full of 'x'?

ShoneAlbert, 2017-12-03:
I met this problem also. and there is no chiplib file. I don not know why .  Is this about version if vcs???
it seems can't load memory from the *.dat file

jwise, 2017-12-07:
I'm closing this, assuming that it's been solved by now, but if not, please feel free to reopen this issue with the requested information.  Thanks!

closed 2017-12-07 by jwise