nvdla/hw #54: Minor issue - tools/bin/tmake shall have execute permission by default

svenka3, 2017-11-19:
Hello,
  I am trying to upgrade to latest version of NVDLA source code as there has been quite a few updates since I tried it in its early days. One thing I noticed is, the "build" process has been introduced - good/bad - I can learn to like that for large project. I follow: http://nvdla.org/integration_guide.html and try and do:

  ./tools/bin/tmake -build vmod

My first trial failed, owing to the "tmake" not having 755 like permission (Am on TCSH if that matters, guess not). It is minor, I had to manually change it, perhaps it is worth considering at your end for the GitHub upload version as well. 

Am now stuck on "can't locate Perl IO/Tee.pm"  issue. Will fight the CPAN later.

Thanks
Srini

jwise, 2017-11-20:
As far as I can tell, it does in fact have execute permissions by default.  `tmake` was introduced in commit c70e7e6; a `git show c70e7e6` shows it was created with execute permissions:

```
jwise@jwise-dt:~/hw$ git show c70e7e6 tools/bin/tmake | head -n8
commit c70e7e69ac72c8bfc45279568120a36f39f4a11a
Author: Eric Zhang <ezhang@nvidia.com>
Date:   Fri Oct 20 01:24:24 2017 -0700

    tools: move bin/etc into tools

diff --git a/tools/bin/tmake b/tools/bin/tmake
new file mode 100755
```

Additionally, GitHub indicates it should be executable: https://github.com/nvdla/hw/blob/master/tools/bin/tmake .

Please check your local config (in particular, your umask); I'll close this for now, but feel free to reopen this issue if you believe there is something that we need to correct.  Thanks for the report!

closed 2017-11-20 by jwise

reanlee, 2018-01-09:
I also met the issue "can't locate Perl IO/Tee.pm"

after installing "libio-tee-perl", build pass!!

sun3388, 2018-01-09:
Hi 
Please try install below Perl lib to see if you are still facing issues. 
sudo apt-get install libyaml-perl
sudo apt-get install libio-tee-perl
if you are still hitting script issues, make sure you have Capture::Tiny and XML::Simple installed.

reanlee, 2018-01-09:
Hi sun,

Thanks for your reply, it's OK now, thanks again :)