nvdla/hw #57: The precision of sigmoid, tanh in SDP

Sonansu04, 2017-11-26:
The SDP uses lookup table to implement the non-linear function like sigmoid, tanh, PReLU.  What's the highest precision it can support? Does it can also support the single/half precision?

kenter83, 2017-11-27:
For sigmoid, tanh, yes, we're using LUT approach. For PReLU, we use fix-function logic to implement (You can see there're <BS|BN|EW>_MUL_PRELU control bit in SDP register).
For your question:
1. The precision for values in LUT entry are: INT16/FP16 for INT/FP pipe respectively; (you can see there're 16bits for S_LUT_ACCESS_DATA.LUT_DATA)
2. The computation precision for INT/FP pipeline are INT32/FP32 respectively;
3. The output  of SDP can be programmed as INT8/INT16/FP16 (controlled by D_DATA_FORMAT.OUT_PRECISION)

jwise, 2017-12-04:
Closing as answered; please feel free to reopen if you have further questions.

closed 2017-12-04 by jwise

Sonansu04, 2017-12-05:
Thank you for the answer! I still have questions about the LUT approach and precision. 
1.The highest precision for the sigmoid, tanh can be up to FP16.  Whether the LUT approach uses direct mapping or function approximation like polynomial approximation? Is the size of the lookup table up to 2^16 for single function for direct mapping?