nvdla/hw #58: about 'Line_stride' and 'Surface_stride' in input_feature_map.dat

huangqiancun, 2017-11-27:
thanks so mush for sharing NVDLA
in the input_feature_map.dat of cc_alexnet_conv5_relu5_int16_dtest_cvsram .
what do these two lines mean？
Line_stride=0x1a0 
Surface_stride=0x1520

zdraw, 2017-11-29:
Comments about how the data is organized. The line_stride means each line is 0x1a0 long so we program the CDMA with that value "write_reg 0xffff1410 0x1a0 #NVDLA_CDMA.D_LINE_STRIDE_0" . To go from surface n to surface n+1, you skip over the lines that make up surface n. This 0x1520 is programmed as "write_reg 0xffff1412 0x1520 #NVDLA_CDMA.D_SURF_STRIDE_0"

huangqiancun, 2017-11-30:
i know the 0x1a0 is programmed as "write_reg 0xffff1410 0x1a0 #NVDLA_CDMA.D_LINE_STRIDE_0" and the  0x1520 is programmed as "write_reg 0xffff1412 0x1520 #NVDLA_CDMA.D_SURF_STRIDE_0".

But what do the Line_stride and Surface_stride really mean,  what really concert me is that what kind of  difference between "Line_stride=0x1a0 Surface_stride=0x1520" and "Line_stride = 0x100 Surface_stride = 0x800" in two difference input_feature.dat.

jwise, 2017-11-30:
We're working on documentation that will describe this in more detail.  But in brief, the line stride, as @zdraw mentioned, is the distance in bytes from one line to the next, and the surface stride is the distance in bytes from one surface to the next.  That is to say, line 0 starts at the base address; line 1 starts at the base address + D_LINE_STRIDE_0; line 2 starts at the base address + D_LINE_STRIDE_0 * 2; etcetera.  It describes the format of the surface in memory.

jwise, 2018-01-08:
We've written some documentation about this: http://nvdla.org/hw/format.html -- please feel free to reopen this bug if you have further questions.  Thanks!

closed 2018-01-08 by jwise