nvdla/hw #60: how should i

huangqiancun, 2017-11-28:


huangqiancun, 2017-11-28:
the register address in the input.txn is differ from the corresponding address in document  Hardware Architectural Specification 
here's an example:
SDP.S_POINTER address: 0xffff2c01 in the input.txn, but 0xe004 in the http://nvdla.org/hwarch.html.
CDMA.S_POINTER address: 0xffff1401 in the input.txn, 0x8004 in the http://nvdla.org/hwarch.html
where can i find the actual relation between those input.txt and document   or how should i config the input.txn according to the document?

zdraw, 2017-11-28:
This was addressed in https://github.com/nvdla/hw/issues/24. The short version is the top 16 bits are control bits and the bottom 16 bits are a shifted version of the address in the HAS.

closed 2017-11-28 by zdraw