nvdla/hw #61: about input.txn's write/read register  command

huangqiancun, 2017-11-29:
i have already know the address from the issue #24 is hardware expects you to have subtracted the global offset and then shifted the result by 2 to get the word address. 

but still, i have a question about and write_reg and read_reg command .
what is the offset and data mean?
input.txn in conv_8x8_fc_int: write_reg 0xffff1402 0x3000f
input.txn in conv_8x8_fc_int: read_reg 0xffff1402 0x000f000f 0x0003000f. 
input.txn in cc_alexnet_conv5_relu5_int16: read_reg 0xffff1402 0x000f000f 0x00010001.
what is the 0x3000f in write_reg and 0x000f000f 0x0003000f 0x000f000f 0x00010001 in read_reg mean? i found that vary from difference input.txn even the same address.

zdraw, 2017-11-29:
If you look at https://github.com/nvdla/doc/blob/master/doc/integration_guide.rst#testbench--traces the fields are a bit clearer.

To me offset is the address offset from the global offset. You could think of it as the address. So "write_reg 0xffff1402 0x3000f" will write 32'h0003000f to csb addr 16'h1402.

For read_reg, one field is a bitmask. This implements a polling read that ends when (read value of csb addr 16'h1402) & 32'h000f000f == 32'h00010001, or when the timeout value is reached if the value never matches. This bitmask was created from the write-only portion of the register. Writes to bits 32'fff0fff0 are dropped and would be read back as 0 so the bitmask is used to ignore those bits when reading.

huangqiancun, 2017-11-30:
Is the 32'h0003000f related to the size of input.feature.dat or input_weight.
I mean weather difference size of input_feature and weight will affect the  offset value 32'h0003000f  or not. why here are 32'h0003000f and somewhere are 0x00010001 etc.
Cause i want to write a new test case use a total difference size of feature and weight, i have to know how to config the input.txn, including the data offset and the address.

jwise, 2017-11-30:
Register 0x1402 is the CDMA arbiter, as seen in the comment:

> write_reg 0xffff1402 0x3000f #NVDLA_CDMA.S_ARBITER_0ARB_WEIGHT=15, ARB_WMB=3

I think we'll have documentation soon coming for how weight compression works, but in the mean time, you can safely use any of the defaults for the arbiter settings.

closed 2017-12-07 by jwise