nvdla/hw #62: while tmake after set tools paths and nv_large, YAML.pm can not been found.

CrazyBingo, 2017-11-30:
#50 @zdraw 
I'm sorry to be a little busy on a business trip, Return this topic back now. As I can't open #50, so set up a nwe one.

YesterdayI have down the code of 20171121, and done as you said:
1: hw path make, input c++/perl/gcc+java path and nv_largd
2: set perl variable in .cshrc
3: then run hw/tools/bin/tmake, but there is sth. wrong tha can not load YAML.pm and interrupted, which is execute： use YAML qw（LoadFile）
@zdraw

the error is like that:
Can't locate YAML.pm in @inc (@inc contains: /usr/local/lib64/perl5 /usr/local/share/perl5 /usr/lib64/perl5/vendor_perl /usr/share/perl5/vendor_perl /usr/lib64/perl5 /usr/share/perl5 .) at ./tmake line 8.
BEGIN failed--compilation aborted at ./tmake line 8.
[image]

seems that perl path variable is wrong,..

jwise, 2017-11-30:
You need to install the YAML module into your local perl installation.  On my Ubuntu machine, this is provided by the package libyaml-perl; if you're running a different distro, you'll need to consult your distro's documentation for how to do that.

I'm closing this for now, expecting that this will solve the problem, but please feel free to reopen it (or open a new issue if you run into a different problem) if the problem persists.  Thanks!

closed 2017-11-30 by jwise

CrazyBingo, 2017-11-30:
Yes, I got it either , and will set up the YAML module, TKS