nvdla/hw #64: About generate netlist of Altera/Xilinx with Synplify, lack of Altera/Xilinx RAM

CrazyBingo, 2017-11-30:
After set Verdi ourself, we will generate netlist of Altera/Xilinx with synplify, is there any Altera/Xilinx RAM for replace, or it must be generated by ourself.

We plan to generate 2 netlist, which is Altera or Xilinx with our FPGA Platform.

Even if don't replace the Sim RAM with Altear/Xilinx, Synply can compile correctly， but will be error while generate the netlist, for the reason there is not enough layout resource. 

So have you generate Altera/Xilinx RAM, or it should be generated by ourself?

jwise, 2017-12-04:
We don't have generated versions of Altera or Xilinx RAMs; indeed, you'll need to build those yourself.  If you build those RAMs, and you're willing to contribute those back, we would greatly appreciate it!

CrazyBingo, 2018-01-19:
2 month ago, I want generate NET of NVDLA in synplify with V2000T, the same as you that the RAM now are only simu RAM, but not Altera or Xilinx RAM， I do not konw when NVDLA can support that, but now  may be we can only replace is by ourself...

peterzh2018888, 2019-01-15:
did you try to create the IP for the nv_small with Vivado? If so, how did you manage the connections with the Zynq UltraScale+?