nvdla/hw #65: Verilog instead of SystemVerilog

elegz, 2017-11-30:
I didn't expect it from Nvidia. Last Verilog standard -- 2001. Merge of SystemVerilog and Verilog standards -- 2005 (12 years ago, it's about 6 generations in microelectronics, like the whole age in our industry). No more Verilog, dead language. SystemVerilog standard's updates -- 2009 & 2012. Any tool supports at least SV-2005. You can even keep the same code, just change all files' extension from *.v to *.sv to allow developers to write more efficient code and contribute back. Shame on you. :) Really, it would be better, if you used VHDL instead of that. :) Such conservatism is the reason why we have been waiting for new SV standard for 5 years and not all tools support last standard or not all features.

fafa1971, 2017-11-30:
I partially agree with your "rant", in particular about declaring modules still using Verilog-1995 style.
But to be fair to NVIDIA, using Verilog ensures total compatibility even for users that don't have access to commercial tools. Take Icarus Verilog and Verilator for example, they are community driven and I doubt they will ever have full support for SystemVerilog.

elegz, 2017-11-30:
May be, I am a maximalist, but, IMHO, it is better to contribute to these free tools than downgrade and slow down everything around. Also I have met a lot of people having access to commercial tools, but still demonstrating such behavior. Don't think it is a real reason. I even thought about HDL generators, but in this case the real source code would be released, not generated one.

jwise, 2017-12-01:
Yeah, it's mostly a tool support issue.  A bunch of our verification environment is SystemVerilog-based, but the standard around here is that for synthesizable code, we use good old plain Verilog.

Part of it is as you mentioned, that some of this was originally generated (though, as noted elsewhere, the generator is no longer the reference: the generated code is, in fact, what we're working with, too).  But part of it is tooling outbound: we're here to build a deep learning accelerator IP, and any time we spend modifying Icarus or Verilator (not to mention the closed source tools that we can't modify!) is time that we are not spending on NVDLA.

I understand the complaint, though, and as we go forward, we will consider using SystemVerilog features where appropriate.

closed 2017-12-01 by jwise

elegz, 2017-12-04:
Oh, initially you used Verilog generator, but C-code is not a reference anymore. It makes some things more clear now. I can imagine how difficult to work with auto-generated code now. Hope that NVIDIA together with comunity will improve it as the project is developed. And I apology if my words sounded rude. I should've started by thanking Nvidia for this code released. :) Good job anyway!