nvdla/hw #66: Failed to build NVDLA !!!

haroonrl, 2017-12-01:
I ran the "./tmake " command to build the NVDLA , i got the error, tried enough to fix it , but unable to do so, please suggest solution !

harry@harry-OptiPlex-3050:~/hw-master/tools/bin$ ./tmake
Can't locate YAML.pm in @INC (you may need to install the YAML module) (@INC contains: /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.22.1 /usr/local/share/perl/5.22.1 /usr/lib/x86_64-linux-gnu/perl5/5.22 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.22 /usr/share/perl/5.22 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base .) at ./tmake line 8.
BEGIN failed--compilation aborted at ./tmake line 8.

jwise, 2017-12-01:
As the error message suggests, you may need to install the YAML Perl module on your system.  On my Ubuntu machine, this is provided by the package libyaml-perl; if you're running a different distro, you'll need to consult your distro's documentation for how to do that. (This is similar to #62.)

I'm closing this for now, expecting that this will solve the problem, but please feel free to reopen it (or open a new issue if you run into a different problem) if the problem persists. Thanks!

closed 2017-12-01 by jwise

wujunning2011, 2017-12-01:
https://github.com/wujunning2011/hw/wiki/Solving-Problems