nvdla/hw #67: Failed to Build NVDLA  ! (even YAML/XS.PM exits)

haroonrl, 2017-12-01:
harry@harry-OptiPlex-3050:~/hw-master/tools/bin$ ./tmake
Can't locate YAML.pm in @INC (you may need to install the YAML module) (@INC contains: /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.22.1 /usr/local/share/perl/5.22.1 /usr/lib/x86_64-linux-gnu/perl5/5.22 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.22 /usr/share/perl/5.22 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base .) at ./tmake line 8.
BEGIN failed--compilation aborted at ./tmake line 8.

then i ran the following command 

harry@harry-OptiPlex-3050:~/hw-master/tools/bin$ **apt-file search YAML/XS.pm**
libyaml-libyaml-perl: /usr/lib/x86_64-linux-gnu/perl5/5.22/YAML/XS.pm

i didn't get why it is doing so.

jwise, 2017-12-01:
`apt-file` tells you what package you need to install, but does not install the package for you.

Also, libyaml-libyaml-perl is different (perversely) than libyaml-perl.  On my system:

```
joshua@nyus:~$ dpkg -S /usr/share/perl5/YAML.pm
libyaml-perl: /usr/share/perl5/YAML.pm
joshua@nyus:~$ dpkg -S /usr/lib/x86_64-linux-gnu/perl5/5.26/YAML/LibYAML.pm
libyaml-libyaml-perl: /usr/lib/x86_64-linux-gnu/perl5/5.26/YAML/LibYAML.pm
```

You need the former, not the latter.  Make sure that libyaml-perl is installed, and also that the file exists on your system.

haroonrl, 2017-12-01:
Now i am gettign the following error ! 

harry@harry-OptiPlex-3050:~/hw-master/tools/bin$ ./tmake
Died at ./tmake line 143.

here is my tmake file :

#!/usr/bin/env perl

##
## build tree
##
use Getopt::Long;
use Data::Dumper;
use YAML qw(LoadFile);
use IO::Tee;
use Carp qw(croak);

my $MAKE = "make";

#=============================
## Options
#=============================
my @build;
my $only;
my $opt_m=""; # optins passed to make
my $log = "build.log";
my $outdir = "outdir";
my $run = 1;
my $debug = 0;
my $project;
my $clobber;
GetOptions (
             "help|h" => sub {usage(); exit 0},
             "build|b=s" => \@build,
             "only=s" => \$only,
             "opt_m=s"   => \$opt_m,
             "log=s"     => \$log,
             "outdir=s"     => \$outdir,
             "project=s" => \$project,
             "run!"      => \$run,
             "debug"     => \$debug,
             "clober|clean"      => \$clobber,
           ) or die "Unrecognized options @ARGV";

push @build, "verif" unless @build;

#=============================
## Global Variables
#=============================
my $cfg = "tools/etc/build.config";
my $outdir_log = "$outdir/$log";
#
#=============================
## MAIN
#=============================
build();

#=============================
## SUB
#=============================
sub build {
    my $outdir_log_tmp = "$outdir/$log".".tmp";
    
    #system ("rm -rf $outdir_log");
    unlink $outdir_log;
    system ("mkdir -p $outdir");
    system ("touch  $outdir_log");
    
    my @projects = $project ? ($project) : get_projects("tree.make");
    foreach my $project (@projects) {
        my $tree = LoadFile($cfg);
        my $done;
        #print Dumper($tree);
        #
        if ($only) {
            croak "[ERROR]: build target $only does not exist in $cfg" unless $tree->{$only};
                
            foreach my $sandbox (@{$tree->{$only}{sandbox}}) {
                system("$MAKE -C $sandbox PROJECT=$project $opt_m | tee $outdir_log_tmp")==0 or die "$sandbox build fail: $?" if $run;
                system ("cat $outdir_log_tmp >> $outdir_log") if $run;
            }
            
        } else {
            while (%$tree) {
                my @keys = keys %$tree;
                
                my $key="unknown";
                foreach my $i (0..$#keys) {
                    #print "key$i=$keys[$i]\n";
                    if (grep {$keys[$i] =~ /^$_/} @build) {
                        $key = $keys[$i];
                        #print "find:key=$key\n";
                        last;
                    }
                }
                # all -build target is built;
                print "key=$key\n" if $debug;
                last if ($key eq "unknown");
    
                #print Dumper($tree);
                
                my $leaf = find_any_leaf($tree,$key,$done);
            
                print "leaf=$leaf\n" if $debug;
            
                foreach my $sandbox (@{$tree->{$leaf}{sandbox}}) {
                    system("$MAKE -C $sandbox PROJECT=$project $opt_m | tee $outdir_log_tmp")==0 or die "$sandbox build fail: $?" if $run;
                    system ("cat $outdir_log_tmp >> $outdir_log") if $run;
                }
                
                $done->{$leaf} = 1;
                delete $tree->{$leaf};
            }
        }
    }
    #system ("rm -rf $outdir_log_tmp");
    unlink $outdir_log_tmp;
    
    print "logfile: $outdir_log\n";
    my $tee = IO::Tee->new(\*STDOUT, $outdir_log);
    print $tee "="x46,"\n";
    print $tee "="x18,"BUILD PASS","="x18,"\n";
    print $tee "="x46,"\n";
}

sub find_any_leaf {
    my $tree = shift;
    my $key = shift;
    my $done = shift;
    
    croak "[ERROR]: dependency $key does not exist in $cfg" unless $tree->{$key};

    my $leaf;
    if (exists $tree->{$key}{dependencies}) {
        #print "find_any_leaf:key=$key\n";
        foreach my $dep (@{$tree->{$key}{dependencies}}) {
            #print "find_any_leaf:dep=$dep\n";
            next if exists $done->{$dep};
            return find_any_leaf($tree,$dep,$done);
        }
    }
    return $key;
}

sub get_projects {
    my $file = shift;

    my @projects;
    open my $fh,"<",$file or die;
    while (my $line = <$fh>) {
        chomp $line;
        next unless $line =~ /^\s*PROJECTS/;
        my $str = $1 if $line =~ /=(.*)/;
        @projects = split(' ',$str);
    }
    close $fh;

    #print Dumper(\@projects);
    return @projects;
}

sub usage()
{
    print
    	"Usage: tmake\n",
}

jwise, 2017-12-01:
You need to run tmake from the top of tree -- from the `hw` directory, run `tools/bin/tmake`.

jwise, 2017-12-07:
I'm closing this, assuming that it has been solved by now, but if this doesn't fix this issue, please feel free to reopen.  Thanks!

closed 2017-12-07 by jwise

aishu319, 2019-10-01:
$perl tmake - build vmod
getting an error while building RTL    

Can't locate YAML.pm in @inc (@inc contains: usr/local/lib64/perl5 /usr/local/share/lib64/perl5 /vendor_perl/usr/share/perl5 /usr/lib64/perl5/usr/share/tmake line 8.
BEGIN failed--compilation aborted at ./tmake line 8.

how to add yaml.pm package in Linux
i'm new to this

prasshantg, 2019-10-01:
@aishu319 please follow below steps to install YAML
```
wget -O YAML-1.24.tar.gz http://search.cpan.org/CPAN/authors/id/T/TI/TINITA/YAML-1.24.tar.gz 
tar -xzvf YAML-1.24.tar.gz 
cd YAML-1.24
perl Makefile.PL
make
sudo make install
wget -O IO-Tee-0.65.tar.gz http://search.cpan.org/CPAN/authors/id/N/NE/NEILB/IO-Tee-0.65.tar.gz 
tar -xzvf IO-Tee-0.65.tar.gz
cd IO-Tee-0.65
perl Makefile.PL
make
sudo make install
cpan -i Capture::Tiny
cpan -i XML::Simple
```

qaziullah, 2021-01-04:
Getting error while building tool/bin/tmake....

$ ./tools/bin/tmake
Can't locate YAML.pm in @INC (you may need to install the YAML module) (@INC contains: /usr/lib/perl5/site_perl /usr/share/perl5/site_perl /usr/lib/perl5/vendor_perl /usr/share/perl5/vendor_perl /usr/lib/perl5/core_perl /usr/share/perl5/core_perl) at ./tools/bin/tmake line 8.
BEGIN failed--compilation aborted at ./tools/bin/tmake line 8.

trying build it in windows. kindly suggest some solutions for windows

APVGITHUB, 2022-07-13:
sudo apt install libconfig-yaml-perl  libcapture-tiny-perl