nvdla/hw #69: cmod build error -- isnan

HKLee2040, 2017-12-01:
‘isnan’ was not declared in this scope

mkdir -p ../outdir/nv_full/cmod/../cmod/hls_wrapper/
/usr/bin/g++ -I./include -I../cmod/include/nvdla_ness_header -I../cmod/include -I../cmod/nvdla_payload -I../cmod/bdma -I../cmod/cacc -I../cmod/cbuf -I../cmod/cdma -I../cmod/cdp -I../cmod/cmac -I../cmod/csb_master -I../cmod/csc -I../cmod/cvif -I../cmod/glb -I../cmod/hls_wrapper -I../cmod/mcif -I../cmod/nvdla_core -I../cmod/pdp -I../cmod/rubik -I../cmod/sdp -I../cmod/bdma/gen -I../cmod/cacc/gen -I../cmod/cbuf/gen -I../cmod/cdma/gen -I../cmod/cdp/gen -I../cmod/cmac/gen -I../cmod/csb_master/gen -I../cmod/csc/gen -I../cmod/cvif/gen -I../cmod/glb/gen -I../cmod/hls_wrapper/gen -I../cmod/mcif/gen -I../cmod/nvdla_core/gen -I../cmod/pdp/gen -I../cmod/rubik/gen -I../cmod/sdp/gen -I../cmod/hls/include -I../cmod/hls/cdma_libs -I../cmod/hls/cdp_libs -I../cmod/hls/csc_libs -I../cmod/hls/sdp -I../cmod/hls/vlibs -I/usr/local/systemc230/include -MMD -MP -fPIC -Wall -Werror -DSC_INCLUDE_DYNAMIC_PROCESSES -Wp,-w -std=c++11  -c ../cmod/hls_wrapper/cdp_hls_wrapper.cpp -o ../outdir/nv_full/cmod/../cmod/hls_wrapper/cdp_hls_wrapper.cpp.o
../cmod/hls_wrapper/cdp_hls_wrapper.cpp: In function ‘void HLS_CDP_lookup_lut(vFp32Type, uint16_t, bool, bool, int16_t*, uint64_t, uint64_t, int16_t, int16_t, int16_t, int8_t, bool&, bool&, bool&, ac_channel<ac_int<17, true> >&)’:
../cmod/hls_wrapper/cdp_hls_wrapper.cpp:251:63: error: ‘isnan’ was not declared in this scope
     if(lut_sub_result_float <= 0 || isnan(lut_sub_result_float))
                                                               ^
../cmod/hls_wrapper/cdp_hls_wrapper.cpp:251:63: note: suggested alternative:
In file included from /usr/local/systemc230/include/systemc.h:45:0,
                 from ./include/log.h:15,
                 from ../cmod/hls_wrapper/cdp_hls_wrapper.cpp:11:
/usr/include/c++/5/cmath:641:5: note:   ‘std::isnan’
     isnan(_Tp __x)
     ^
../cmod/hls_wrapper/cdp_hls_wrapper.cpp: In function ‘void HLS_CDP_lookup_fp16(int16_t*, int16_t*, int16_t*, uint16_t, uint16_t, bool, bool, bool, uint64_t, int16_t, int8_t, uint64_t, uint64_t, int8_t, uint64_t, uint16_t, uint16_t, uint8_t, uint32_t, uint16_t, uint8_t, int16_t, int16_t, int16_t, int16_t, bool, bool, int16_t*, uint32_t*, uint32_t*, uint32_t*, uint32_t*, uint32_t*)’:
../cmod/hls_wrapper/cdp_hls_wrapper.cpp:660:41: error: ‘isnan’ was not declared in this scope
             if(isnan(icvt_data_out_float))
                                         ^
../cmod/hls_wrapper/cdp_hls_wrapper.cpp:660:41: note: suggested alternative:
In file included from /usr/local/systemc230/include/systemc.h:45:0,
                 from ./include/log.h:15,
                 from ../cmod/hls_wrapper/cdp_hls_wrapper.cpp:11:
/usr/include/c++/5/cmath:641:5: note:   ‘std::isnan’
     isnan(_Tp __x)
     ^
make: *** [../outdir/nv_full/cmod/../cmod/hls_wrapper/cdp_hls_wrapper.cpp.o] Error 1
Makefile:165: recipe for target '../outdir/nv_full/cmod/../cmod/hls_wrapper/cdp_hls_wrapper.cpp.o' failed

jwise, 2017-12-02:
Hmm, what version of gcc are you using?  Please note that we've tested with a pretty old version of gcc, sadly; if there are compatibility issues with newer versions, I'd like to hear about it, though.

HKLee2040, 2017-12-02:
The version of gcc I used is 5.4.0.

jwise, 2017-12-02:
Try backing down to 4.x.x, and seeing if that works?

We haven't tested with 5.x.x and above.  But it is possible that making the change that it suggests -- "std::isnan" instead of "isnan" might work.  I'm not sure where that lies in HLS tooling support, though.

HKLee2040, 2017-12-02:
It's still not working when I use 4.8.5.

But, if I replace 'isnan' with 'std::isnan' under 4.8.5, the building process is done.

jarodw0723, 2017-12-04:
What is your glibc version (ldd --version will show)? With glibc >= 2.23 this could be an issue, and replace 'isnan' with 'std::isnan' indeed will fix it.

HKLee2040, 2017-12-04:
The version of glibc is 2.23.

jwise, 2017-12-04:
@jarodw0723 Can we confirm this won't break the HLS build if we make this change to the cmod?

@HKLee2040 Since I can't replicate the issue on my machine, would you mind submitting a PR with all the right std::isnan changes to make the build pass for you?  If that doesn't break the HLS build, we'll take that change.  Thanks!

jarodw0723, 2017-12-05:
I don't think it will break anything since they are essentially the same thing. But better confirm with HW team on this.

HKLee2040, 2017-12-05:
@jwise  Actually, I'm new here.  I have no idea to submit a PR.

jwise, 2017-12-07:
Awaiting merge of PR #74.  (I'm out of the office right now but I'll merge this when I get back next week).  Thanks for your help, @HKLee2040

jwise, 2017-12-13:
Fixed in 0db9bac.  Thanks again!

closed 2017-12-13 by jwise