nvdla/hw #72: Failed to build verilator

HKLee2040, 2017-12-02:
Verilator 3.912
clang 3.8.0
ubuntu 16.04

"Unsupported or unknown PLI call: $sformatf"

==============================================
files are generated under /media/rogers/DATA/work/nvdla_org/trunk/outdir/nv_full/vmod/nvdla/top
==============================================
make: Leaving directory '/media/rogers/DATA/work/nvdla_org/trunk/vmod/nvdla/top'
make: Entering directory '/media/rogers/DATA/work/nvdla_org/trunk/verif/verilator'
/usr/bin/verilator --cc --exe -f verilator.f --Mdir ../../outdir/nv_full/verilator/ nvdla.cpp --compiler clang --output-split 250000000
%Error: ../../outdir/nv_full/vmod/vlibs/RANDFUNC.vlib:52: Unsupported or unknown PLI call: $sformatf
%Error: Exiting due to 1 error(s)
%Error: See the manual and http://www.veripool.org/verilator for more assistance.
%Error: Command Failed /usr/bin/verilator_bin --cc --exe -f verilator.f --Mdir ../../outdir/nv_full/verilator/ nvdla.cpp --compiler clang --output-split 250000000
make: *** [../../outdir/nv_full/verilator/VNV_nvdla.mk] Error 10
Makefile:21: recipe for target '../../outdir/nv_full/verilator/VNV_nvdla.mk' failed
make: Leaving directory '/media/rogers/DATA/work/nvdla_org/trunk/verif/verilator'
logfile: outdir/build.log

HKLee2040, 2017-12-05:
/usr/bin/verilator_bin --version
==> 3.874

After remove the 3.874 version, it can work now.

jwise, 2017-12-05:
Great!  Glad you got it working.  I'll close this bug, but if you have further problems with this, please feel free to reopen it.

closed 2017-12-05 by jwise

koroyuki, 2018-03-07:
Dear Sir,
This is my first time to write a comment here.
I have the same issue with HKLee2040 of Dec.5, 2017.
My Verilator version is  3.874-1.
HKLee2040 wrote that "After remove the 3.874 version, it can work now."
May I know the detail commands that HKLee2040 did?
Thanks in advance. Tago

HKLee2040, 2018-03-07:
@koroyuki Sorry, I do not remember the detail commands, it's several months ago.
I think it should be something like:
sudo apt-get remove verilator
or
sudo apt-get remove --auto-remove verilator
Then install Verilator 3.912.

I did not try docker.  Maybe you can try it.

BTW, I do not use verilator because it needs large DDR and my computer can't afford it.

koroyuki, 2018-03-07:
Dear HKL332040
Thank you very much for your quick reply.
I understand the remove step.
Is the installation command that 
  sudo apt-get install Verilator=3.912
?

Tago

HKLee2040, 2018-03-07:
@koroyuki 
As my remember, I download the 3.912 source code, build the code, and then make install.

https://www.veripool.org/projects/verilator/wiki/Installing

koroyuki, 2018-03-07:
Dear HKLee2040
Thank you very much for your suggestion again.
Before I read your suggestion above, I did "sudo apt-get remove verilator", and then
:~$ sudo apt-get install verilator
Reading package lists... Done
Building dependency tree       
Reading state information... Done
verilator is already the newest version (3.874-1).
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.

So 3.912 was not obtained. Only version 3.874-1 was installed.
I try tar ball method under the URL you gave me now on.

 3.874-1.

koroyuki, 2018-03-07:
Dear HKLee2040

I followed 
https://www.veripool.org/projects/verilator/wiki/Installing
to build Verilator 3.916 environment.
The build looked success!
How can I do a sanity check next?

HKLee2040, 2018-03-07:
@koroyuki 

http://nvdla.org/integration_guide.html#verilator-testbench

koroyuki, 2018-03-13:
Dear HKLee2040,
Thank you very much for your suggestion and sorry for my slow response.
According to Verilator Testbench in nvdla integration guide, I tyoed
./tools/bin/tmake -build verilator
*** build pass *** was displayed.
Next, I typed
make run TEST_sanity0

After several ten minute run with lots of warning messages, It ended with
make: *** [../../outdir/nv_full/verilator/VNV_nvdla.mk] Error 255

I wildly guess it is maybe due to out of memory. My PC has 16GB memory.
I am going to configure nvdla RTL with "nv_small" as shown in Tree Build section
 in the integration guide.
Your suggestion is very welome!
koroyuki

HKLee2040, 2018-03-14:
@koroyuki 
I can't reopen this issue.  I suggest that you can open a new issue to discuss this problem.

koroyuki, 2018-03-14:
Dear HKLee2040,
Thank you very much for your strong support.
I found my mistakes for systemc-2.3.0 path and verilator path in make execution.
I am trying to build "nv_large nv_small" environment instead of "nv_full". 
I typed the following commands.
>git clone https://github.com/nvdla.hw.git nvdla_hw
>cd nvdla_hw
>git checkout master    // checkout before 2017 Nov. version
>make
I am asked cpp path, gcc path, java path, and  perl path only.
I am not asked systemc path, verilator path and clang path.
For "make" of "nv_full" case, the all paths above are asked.
make ends with *** pass ***. 
However, ./tools/bin/tmake -build vmod failed.
Do you have a suggestion?
Best regards
koroyuki

HKLee2040, 2018-03-16:
@koroyuki 
I'm afraid that nv_small or nv_large is not ready.
https://github.com/nvdla/hw/issues/94

koroyuki, 2018-03-19:
Dear Heng-Kuan Lee.
Thank you very much for your support.
I see. I use the latest "nv_full".
koroyuki

koroyuki, 2018-03-22:
Dear HKLee2040

I installed verilator 3.923 from veripool and make it. I followed
http://nvdla.org/integration_guide.html#verilator-testbench.
I still face a "failed" as shown below. Would you have a suggestion?

tago@d200:~/nvdla_hw$ ./tools/bin/tmake -build verilator
make: Entering directory '/home/tago/nvdla_hw/spec/manual'
/usr/bin/java -jar Ordt.jar -parms test.parms -systemverilog /home/tago/nvdla_hw/outdir/nv_full/spec/manual/sv/ -verilog /home/tago/nvdla_hw/outdir/nv_full/spec/manual/regs_v.v -uvmregs /home/tago/nvdla_hw/outdir/nv_full/spec/manual/regs_ral.sv -cppmod /home/tago/nvdla_hw/outdir/nv_full/spec/manual/cmod -cppdrvmod /home/tago/nvdla_hw/outdir/nv_full/spec/manual/dmod test.rdl
Open Register Design Tool, version=170915.01, input=test.rdl
--- omitted ---
==============================================
files are generated under /home/tago/nvdla_hw/outdir/nv_full/vmod/nvdla/top
==============================================
make: Leaving directory '/home/tago/nvdla_hw/vmod/nvdla/top'
make: Entering directory '/home/tago/nvdla_hw/verif/verilator'
rm -f ../../outdir/nv_full/verilator/nvdla.cpp
cp nvdla.cpp ../../outdir/nv_full/verilator/nvdla.cpp
make -C ../../outdir/nv_full/verilator -f VNV_nvdla.mk CC=/usr/local/clang CXX=/usr/local/clang++ VM_PARALLEL_BUILDS=1
VNV_nvdla.mk:53: /home/kocha/tools/verilator/share/verilator/include/verilated.mk: No such file or directory
make[1]: Entering directory '/home/tago/nvdla_hw/outdir/nv_full/verilator'
make[1]: *** No rule to make target '/home/kocha/tools/verilator/share/verilator/include/verilated.mk'.  Stop.
make[1]: Leaving directory '/home/tago/nvdla_hw/outdir/nv_full/verilator'
make: *** [../../outdir/nv_full/verilator/VNV_nvdla] Error 2
Makefile:24: recipe for target '../../outdir/nv_full/verilator/VNV_nvdla' failed
make: Leaving directory '/home/tago/nvdla_hw/verif/verilator'
logfile: outdir/build.log
==============================================
==================BUILD PASS==================
==============================================
tago@d200:~/nvdla_hw$ ls outdir/build.log
outdir/build.log

Best regards
koroyuki