nvdla/hw #73: About dump_mem

ShoneAlbert, 2017-12-03:
I can compile all the files without any errors. but is hint me can't fine dump_mem file and tells me there is no ...0.chilp.raw2 or something. And I checked it again find if it exists the command load_mem. it will occurred this kind of error. and i want to know the reason. 
Thank you!!!!!!

jwise, 2017-12-04:
This is not enough information to diagnose the issue; a distillation of error messages is not sufficient, but instead we'll need the exact errors that are produced.  Can you please [create a gist](https://gist.github.com/) with a full log demonstrating the problem?

Thanks!

ShoneAlbert, 2017-12-05:
And I figure out this problem. The reason is the version of the software. I changed the synopsys software to 2016. And then can run . I find the same prolems before.  And you mentioned it the mem file is all XXXX. anyway thank you

closed 2017-12-05 by ShoneAlbert