nvdla/hw #75: how to use libnvdla_cmod.so

wuchunxuan, 2017-12-07:
hi all:
    i can build c mode,  but i donot know how to use libnvdla_cmod.so
can you have example to use libnvdla_cmod.so

jwise, 2017-12-07:
We're working on a Virtual Platform release that will have an example of integrating it into Qbox.  The ETA on that is a few days to a week.

wuchunxuan, 2017-12-07:
OK, i wait it

jwise, 2017-12-12:
The libnvdla_cmod.so exposes a SystemC module, "NV_nvdla".  In the recently release Virtual Platform tree we have an example of instantiating one, and connecting it to other SystemC buses:

https://github.com/nvdla/vp/blob/master/models/nvdla/src/nvdla_top.cpp

This is a relatively complicated example; unfortunately, we don't have something simpler right now, but hopefully this will get you started. I'm going to close this issue for now, but please feel free to reopen it if you have further questions!

closed 2017-12-12 by jwise

wuchunxuan, 2017-12-13:
i think this is too hard,   why have no simple example  with 'main'

wuchunxuan, 2017-12-13:
Hi Subhash -- yesterday, we released the software stack to run on top of NVDLA, along with the Virtual Platform, which should provide some examples and an environment in which you can do further development. Have a look at that?  where it is??

sun3388, 2017-12-13:
Please refer to this doc: http://nvdla.org/vp.html, 
it has virtual platform and sw setup info, 
and you don't really care how to leverage the single so lib, 
you can boot OS and SW tests directly on this virtual platform following the steps in the doc.
related repo : 
https://github.com/nvdla/vp
https://github.com/nvdla/sw

ShoneAlbert, 2017-12-14:
I want to ask how to use the SW 
I have installed the Qbox .