nvdla/hw #76: Where is the HW_CONFIGx of GLB registers in NVDLA verilog code?

lishen565, 2017-12-07:
The register HW_CONFIG0 and HW_CONFIG1 HW_CONFIG2 of GLB registers are important for me to understand the RTL code and config NVDLA, but I can't find these code in NVDLA verilog code. Will these code of the  HW_CONFIGx  be updated and added in the future?

jwise, 2017-12-07:
We're working on a revamp of the capabilities scheme.  Since the capabilities of the nvdlav1 branch will be more or less fixed, we'll remove that from the documentation; hopefully, we'll have some discussion of how to specify capabilities when we're ready to share more about the configurable release.

lishen565, 2017-12-10:
Thank you for your answer.The information on  http://nvdla.org/  or the RTL code annotation are both too simple for me to understand the function of each signal inside NVDLA. Is there a more detailed document about the NVDLA‘s signal line ?

ShoneAlbert, 2017-12-10:
You can't edit hw-config2 to change the MAC_Arrays .  You should set a Mac array size before. the spec file can give you some information. And hope you can go through it . I just fix this problems. if you have any questions you can ask me . I will help you as far as I can. Because this difficult problems is too hard , I hope I can make friends to read and understand it together.
and hw-nvdla1 can't modify