nvdla/hw #77: build verilator

HKLee2040, 2017-12-08:
There are so many Width mismatch warnings shown as below, and the building process is stuck at last line.  Does anyone else have the same problem?

%Warning-WIDTH: ../../outdir/nv_full/vmod/nvdla/csb_master/NV_NVDLA_csb_master.v:1971: Operator ASSIGNDLY expects 34 bits on the Assign RHS, but Assign RHS's CONST '?32?bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx' generates 32 bits.
%Warning-WIDTH: ../../outdir/nv_full/vmod/nvdla/csb_master/NV_NVDLA_csb_master.v:1988: Operator ASSIGNDLY expects 34 bits on the Assign RHS, but Assign RHS's CONST '?32?bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx' generates 32 bits.
%Warning-WIDTH: ../../outdir/nv_full/vmod/nvdla/csb_master/NV_NVDLA_csb_master.v:2005: Operator ASSIGNDLY expects 34 bits on the Assign RHS, but Assign RHS's CONST '?32?bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx' generates 32 bits.
%Warning-UNSIGNED: ../../outdir/nv_full/vmod/nvdla/pdp/NV_NVDLA_PDP_CORE_cal2d.v:2331: Comparison is constant due to unsigned arithmetic
%Warning-UNSIGNED: ../../outdir/nv_full/vmod/nvdla/pdp/NV_NVDLA_PDP_CORE_cal1d.v:2124: Comparison is constant due to unsigned arithmetic
%Warning-UNSIGNED: ../../outdir/nv_full/vmod/nvdla/sdp/NV_NVDLA_SDP_core.v:1081: Comparison is constant due to unsigned arithmetic
%Warning-CMPCONST: ../../outdir/nv_full/vmod/nvdla/cacc/NV_NVDLA_CACC_delivery_ctrl.v:1451: Comparison is constant due to limited range
%Warning-UNSIGNED: ../../outdir/nv_full/vmod/nvdla/cdp/NV_NVDLA_CDP_wdma.v:750: Comparison is constant due to unsigned arithmetic
%Warning-UNSIGNED: ../../outdir/nv_full/vmod/nvdla/cdp/NV_NVDLA_CDP_wdma.v:810: Comparison is constant due to unsigned arithmetic

jwise, 2017-12-08:
There are a lot of warnings but after you get those messages the build should still be running (though it will not print progress).  Please make sure that you haven't run out of memory -- if you've hit swap, then the build will slow down an enormous amount.  On my machine, the build needs about 20GB of RAM, and takes about 7 minutes.  If you're building on a machine with 24GB or more, and it's taking longer than half an hour or so, please let me know your system specs and version of verilator, and perhaps output of 'free' while running the build?

HKLee2040, 2017-12-08:
OK, I know where the problem is.  My machine is notebook, only 4GB DRAM.  I think I need to change the machine.

jwise, 2017-12-09:
Ok, I'll close this for now.  But if you continue to have problems on a machine with sufficient RAM, please feel free to reopen this bug.  Thanks!

closed 2017-12-09 by jwise