nvdla/hw #79: how to use cmod

ShoneAlbert, 2017-12-14:
I have installed QBOX. But  I don't konw how to combine with SW

prasshantg, 2017-12-14:
After you launch cmod using "aarch64_toplevel -c aarch64_nvdla.lua"

1. Either use prebuilt binaries for SW from https://github.com/nvdla/sw/tree/master/prebuilt/linux or build new binaries as instructed in https://github.com/nvdla/sw
2. Use https://github.com/nvdla/sw/blob/master/scripts/init_dla.sh to install drivers
3. After that you can run test applications, usage guide for test app http://nvdla.org/sw/test_application.html

More SW documentation http://nvdla.org/sw/contents.html

HKLee2040, 2017-12-16:
After launch cmod using "aarch64_toplevel -c aarch64_nvdla.lua" with prebuilt binaries, how can I install drivers?  
After "mount -t 9p -o trans=virtio r /mnt"
I can't find drm.ko and opendla.ko in /mnt.
What do I miss?

jwise, 2018-01-08:
Looks like Prashant answered your question, so I'll close this -- feel free to reopen if you have further issues with this.

closed 2018-01-08 by jwise