nvdla/hw #82: NVDLA Create IP and Package (using Xilinx Vivado)

haroonrl, 2017-12-24:
Hi all, 
I am trying (NVDLA )to create and package IP , but i am got the following warnings: 

[IP_Flow 19-3153] Bus Interface 'dla_core_clk': ASSOCIATED_BUSIF bus parameter is missing.
 
why is it so, please help to figure out.

jwise, 2018-01-08:
We don't have experience with Vivado, but please let us know if you find out what's going on here!  I'll leave this open as a tracking issue for Vivado.

jevogel, 2018-01-24:
Check out this thread on the Xilinx forums: [Safe to ignore missing ASSOCIATED_BUSIF bus parameter](https://forums.xilinx.com/t5/Design-Entry/Safe-to-ignore-missing-ASSOCIATED-BUSIF-bus-parameter-in-2014-4/m-p/602657/highlight/true#M8563).

If there are no bus interfaces used, you can ignore this parameter. Otherwise if there is a bus interface associated with this clock you should enter it.

killerzula, 2019-10-31:
SO @haroonrl , did you solve the issue? If yes, then how?