nvdla/hw #84: Error try to generate DEF files for cmod_top

lvcargnini, 2018-01-11:
Hello,

I observed the following issues:
with GCC/LD I got the following error:
make: Entering directory '/wada/nvidia/hw/spec/defs'
/usr/bin/g++ -undef -nostdinc -P -C nv_full.spec -o /wada/nvidia/hw/outdir/nv_full/spec/defs/project.def
nv_full.spec: file not recognized: File format not recognized
collect2: error: ld returned 1 exit status
make: *** [/wada/nvidia/hw/outdir/nv_full/spec/defs/project.def] Error 1

Using LLVM/LD:
/usr/bin/clang++ -undef -nostdinc -P -C nv_full.spec -o /wada/nvidia/hw/outdir/nv_full/spec/defs/project.def
clang: warning: argument unused during compilation: '-undef' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-nostdinc' [-Wunused-command-line-argument]
/usr/bin/ld: error: undefined symbol: main

jeremychritz, 2018-01-12:
I'm not apart of the NVDLA team, but had the same issue, check out #52 .
>You probably shouldn't use g++, but should use cpp, which is for pre-processing. Change your tree.make 'CPP := /usr/local/bin/cpp'

Making that change to my tree.make solved the issue.

jwise, 2018-01-23:
Closing this issue, since @jeremychritz's solution is in fact correct.  If you make that change and still have problems, please feel free to reopen this issue.

closed 2018-01-23 by jwise