nvdla/hw #86: I have configured an nv_small model，but the DBBIF data width is not the 64-bit; I have defined the "NVDLA_PRIMARY_MEMIF_WIDTH 64"(as follow picture), but it

ckhm, 2018-01-16:


wujunning2011, 2018-01-16:
please check the MAC array's param and result, maybe they are the same as nv_lagre.

ckhm, 2018-01-16:
@wujunning2011  Do you means that"master/spec/defs/project.spec"??? I have changed the project.spec 
according to the nv_small.spec

stevewch, 2018-01-19:
I also tried same configuration (nv_small).  "MAC_ATOMIC_C_SIZE" and "MAC_ATOMIC_K_SIZE" are reduced to 8; bus interface data width is still 512 bit, same as original post described.  Also multiplier array is same size as nv_full.
From document it states current nvldav1 uses nv_full only.  Hope next release will have both multiply array and bus width configurable via spec file??

tariqafzal, 2018-01-22:
If anyone has nv_small configuration working(JunningWu?) then can they publish their nv_small.spec along with other changes.  The changes need not to be fully functional.  My interests are exploratory in nature at this time.

jwise, 2018-01-23:
Currently, only `nv_full` is supported.  The parameters are informational, but do not change anything in the RTL build.  (The GitHub `master` branch has been updated to only have one configuration, to reflect this reality.)

The configurable release is on the horizon.

JunningWu, 2018-01-23:
what my guess is verified!!! sigh~~~~
Hoping for future release....

closed 2018-01-27 by ckhm