nvdla/hw #87: ./tools/bin/tmake -build verif_sim failed

blueardour, 2018-01-17:
Hi, have anyone ever run './tools/bin/tmake -build verif_sim' successfully?

After change the vcs tool path according to my machine, it failed to run the command. build.log is here:

make: Entering directory `/workspace/git/nvidia-hw/verif/sim'
DUTDIR = dut
VIPDIR = vip
Using cleartext directories
/workspace/soft/vcs/K-2015.09-SP2-1/bin/vcs -f /workspace/git/nvidia-hw/verif/dut/dut.f  -Xddg=0x1 -o  -cpp   +nospecify +notimingchecks +define+VLIB_NO_UDP +warn=noTFIPC +warn=noTMR -full64 -y /workspace/git/nvidia-hw/verif/synth_tb -y /workspace/git/nvidia-hw/verif/../outdir/nv_full/vmod/vlibs +incdir+/workspace/git/nvidia-hw/verif/synth_tb +incdir+/workspace/git/nvidia-hw/verif/dut +incdir+/workspace/git/nvidia-hw/verif/../outdir/nv_full/vmod/vlibs +incdir+/workspace/git/nvidia-hw/verif/../outdir/nv_full/vmod/include +incdir+/workspace/git/nvidia-hw/verif/../outdir/nv_full/vmod/vlibs +incdir+..  +define+DESIGNWARE_NOEXIST +vcs+lic+wait -sverilog +libext+.v +libext+.sv -timescale=1ns/1ns -debug_all +define+NVTOOLS_SYNC2D_GENERIC_CELL +define+NO_PERFMON_HISTOGRAM +define+PRAND_OFF  +define+NO_DUMPS /workspace/git/nvidia-hw/verif/synth_tb/tb_top.v /workspace/git/nvidia-hw/verif/synth_tb/csb_master.v /workspace/git/nvidia-hw/verif/synth_tb/csb_master_seq.v /workspace/git/nvidia-hw/verif/synth_tb/axi_slave.v /workspace/git/nvidia-hw/verif/synth_tb/id_fifo.v /workspace/git/nvidia-hw/verif/synth_tb/memory.v /workspace/git/nvidia-hw/verif/synth_tb/memresp_fifo.v /workspace/git/nvidia-hw/verif/synth_tb/raddr_fifo.v /workspace/git/nvidia-hw/verif/synth_tb/slave_mem_wrap.v /workspace/git/nvidia-hw/verif/synth_tb/waddr_fifo.v /workspace/git/nvidia-hw/verif/synth_tb/wdata_fifo.v /workspace/git/nvidia-hw/verif/synth_tb/wstrb_fifo.v /workspace/git/nvidia-hw/verif/synth_tb/clk_divider.v /workspace/git/nvidia-hw/verif/synth_tb/slave2mem_rd.v /workspace/git/nvidia-hw/verif/synth_tb/slave2mem_wr.v /workspace/git/nvidia-hw/verif/../outdir/nv_full/vmod/vlibs/NV_DW02_tree.v /workspace/git/nvidia-hw/verif/../outdir/nv_full/vmod/vlibs/NV_DW_lsd.v /workspace/git/nvidia-hw/verif/../outdir/nv_full/vmod/vlibs/NV_DW_minmax.v -l .compile.log    -Mdir=Csrc ; ./checkcompile.pl .compile.log 

Warning-[LINX_KRNL] Unsupported Linux kernel
  Linux kernel '4.4.0-31-generic' is not supported.
  Supported versions are 2.4* or 2.6*.

                         Chronologic VCS (TM)
      Version K-2015.09-SP2-1_Full64 -- Tue Jan 16 22:20:27 2018
               Copyright (c) 1991-2015 by Synopsys Inc.
                         ALL RIGHTS RESERVED

This program is proprietary and confidential information of Synopsys Inc.
and may be used and disclosed only as authorized in a license agreement
controlling such use and disclosure.

CPU time: .868 seconds to compile

checkcompile : Compile FAILED (Warnings=0, Errors=0, ExecutableReady=0) 
The executable () has been removed.

#TESTDIR      := ../traces/traceplayer/sanity0
#FULLTESTDIR  := /workspace/git/nvidia-hw/verif/sim/../traces/traceplayer/sanity0
#TESTNAME     := sanity0
#SIMTESTDIR   := /workspace/git/nvidia-hw/verif/sim/sanity0
/bin/mkdir -p sanity0 ; \
	/bin/cp -Trf /workspace/git/nvidia-hw/verif/sim/../traces/traceplayer/sanity0 /workspace/git/nvidia-hw/verif/sim/sanity0 ; \
	cd /workspace/git/nvidia-hw/verif/sim/sanity0 ; \
	/bin/ln -fs /workspace/git/nvidia-hw/verif/sim/ ; \
	/bin/ln -fs /workspace/git/nvidia-hw/verif/sim/.daidir ; \
	/bin/rm -f input.txn.raw ; \
	/bin/rm -f *.raw2 ; \
	/bin/rm -f test.log ; \
	/workspace/git/nvidia-hw/verif/synth_tb/sim_scripts/slave_mem.cfg.pl -outFile ./slave_mem.cfg  +continue_on_fail ; \
	/workspace/git/nvidia-hw/verif/synth_tb/sim_scripts/inp_txn_to_hexdump.pl  /workspace/git/nvidia-hw/verif/sim/sanity0 ; \
	  -l test.log  +read_reg_poll_retries=10 +continue_on_fail +continue_on_fail   +input_file=/workspace/git/nvidia-hw/verif/sim/sanity0/input.txn +input_dir=/workspace/git/nvidia-hw/verif/sim/sanity0 ;
Running /workspace/git/nvidia-hw/verif/synth_tb/sim_scripts/slave_mem.cfg.pl
default 32 memory width
accessing /workspace/git/nvidia-hw/verif/sim/sanity0/input.txn
make: Leaving directory `/workspace/git/nvidia-hw/verif/sim'

====================
seemed to compile failed with no warning or errors.

jwise, 2018-01-23:
This should work.  I wonder why VCS isn't completing.  Can you run the VCS command by hand and see what happens?

jwise, 2018-02-06:
Having no update since, I'm going to close this issue; please feel free to reopen it (or comment on it, if you're not able to reopen it directly) if you continue to have issues.  Thanks!

closed 2018-02-06 by jwise

CrazyBingo, 2018-02-26:
Warning-[LINX_KRNL] Unsupported Linux kernel
Linux kernel '4.4.0-31-generic' is not supported.
Supported versions are 2.4* or 2.6*.

I have this warming of VCS yet, do not know if it will make failed while generate sanity simulations.