nvdla/hw #88: nv_full: #MACs and precision

ratnayak, 2018-01-17:
1) its stated that nv_full is multi-precision. does it mean we can program the precision selection--int8b, int16n or fp16b--after HW implementation?

2) also its stated that nv_full has 2048 MACs (precision not specified). if run under fp16b does it has 2048 or 1024 fp16b MACs?

thanks
Ruwan

jwise, 2018-01-23:
Yes, this is programmable at runtime -- I believe you get 2048 MACs, no matter what precision you select.

jwise, 2018-01-23:
(Closing this issue for now -- feel free to reopen if needed.)

closed 2018-01-23 by jwise

ratnayak, 2018-01-24:
there is only 1024 instentiations of NV_NVDLA_CMAC_CORE_MAC_mul module for nv_full. 
so same question but perhaps in a different form: A single instentiation of NV_NVDLA_CMAC_CORE_MAC_mul module in nv_full--can it be programmed to do two int8b-MACs or a single int16b-MACs based on the reg2dp_proc_precision setting?
Thanks.