nvdla/hw #90: NVDLA running on FPGA

ratnayak, 2018-01-24:
1) Has nv_full (or nv_large or nv_small)  been run sucessfuly in a stand alone FPGA? If so which device was it run on? (we have tried fitting this into to an FPGA w an ARM processor and current largest Xilinx FPGA—Zynq ZU19EG cannot accommodate nv_full).

2)      Has nv_full (or nv_large or nv_small) been run sucessfuly on Amazon EC2-F1 as suggested in the website? nv_full is unlikely to fit in a single FPGA (which are Xilinx UltSc+ Vertex VU9P) availble in EC2-F1. If nv_full was run in EC2 were they distributed over serveral FPGAs?

JunningWu, 2018-01-25:
LUTs is more than 95% for 2000t !!!!!!
We are planing to split the NVDLA_TOP to some sub-modules. 
May it be work.

jwise, 2018-02-06:
I believe that we have nv_small running on EC2-F1 internally, but we're still working out the rest of the kinks with the remainder of the nv_small release.  (As you noted, nv_full is too large to run on VU9P.)

If you intend to run nv_full on FPGAs, we recommend partitioning it at the top level, as @JunningWu notes.  The unit-to-unit interfaces should be friendly for that kind of partitioning (given that they are all FIFO-driven).  We would be interested in experience reports from that!

In the mean time, I'll mark this 'waiting for future release', since this depends on the nv_small release, coming soon.

CrazyBingo, 2018-02-08:
@jwise  So as you said, you will run nv_small on EC2-F1 FPGA platform， which FPGA is VU9P(258,6,150 Logic cells), can you tell us that how much the area will  

Second, if we partition the NV_FULL from the top level, we will need FPGAs which have enough interface between each others, maye be it is a problem.

ratnayak, 2018-03-28:
This is an inquiry to the wider audience who are working on getting NVDLA running on a FPGA platform--we'd like to share what we are doing and check progress on other groups out there . 

we are putting nv_full (as nv_small is not available as yet) on Xilinx VU440 (largest FPGA by Xilinx). nv_full fits comfortably on VU440 utilizing about 82% LUTs as per Vivado implementation. 
we'd have the host running on an ARM A53 processor on a Xilinx Zynq (ASIC ARM). the signalling between two FPGAs--Zynq(host) and VU440(nv_full) will be via direct parallel connections (simple & straightforward) and we have enough IOs for this.

a) we'd like to know if anyone out there has already implemented nv_full on a FPGA platform and successful in getting it running as expected.
b) we would very much like to sync w you--either you got it running or still some distance to go-- perhaps we can discuss details, inputs on issues encountered, how resolved them etc. 

look forward to hearing from you.

killerzula, 2019-09-19:
Try using only the fpga rams in `hw/outdir/nv_small/vmod/rams/fpga` i.e. remove the folder `synth` and `model` in the specified directory. I have attached by utilization report for zedboard zynq 7000 series Dev kit. 

> Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.
-----------------------------------------------------------------------------------------------------------
| Tool Version : Vivado v.2018.3 (win64) Build 2405991 Thu Dec  6 23:38:27 MST 2018
| Date         : Wed Sep 18 12:58:25 2019
| Host         : HP-PC running 64-bit Service Pack 1  (build 7601)
| Command      : report_utilization -file NV_nvdla_utilization_synth.rpt -pb NV_nvdla_utilization_synth.pb
| Design       : NV_nvdla
| Device       : 7z020clg484-1
| Design State : Synthesized
-----------------------------------------------------------------------------------------------------------

Utilization Design Information

Table of Contents
-----------------
1. Slice Logic
1.1 Summary of Registers by Type
2. Memory
3. DSP
4. IO and GT Specific
5. Clocking
6. Specific Feature
7. Primitives
8. Black Boxes
9. Instantiated Netlists

1. Slice Logic
--------------

+----------------------------+-------+-------+-----------+--------+
|          Site Type         |  Used | Fixed | Available |  Util% |
+----------------------------+-------+-------+-----------+--------+
| Slice LUTs*                | 76280 |     0 |     53200 | 143.38 |
|   LUT as Logic             | 76084 |     0 |     53200 | 143.02 |
|   LUT as Memory            |   196 |     0 |     17400 |   1.13 |
|     LUT as Distributed RAM |   188 |     0 |           |        |
|     LUT as Shift Register  |     8 |     0 |           |        |
| Slice Registers            | 80910 |     0 |    106400 |  76.04 |
|   Register as Flip Flop    | 80910 |     0 |    106400 |  76.04 |
|   Register as Latch        |     0 |     0 |    106400 |   0.00 |
| F7 Muxes                   |  3201 |     0 |     26600 |  12.03 |
| F8 Muxes                   |  1118 |     0 |     13300 |   8.41 |
+----------------------------+-------+-------+-----------+--------+
* Warning! The Final LUT count, after physical optimizations and full implementation, is typically lower. Run opt_design after synthesis, if not already completed, for a more realistic count.

1.1 Summary of Registers by Type
--------------------------------

+-------+--------------+-------------+--------------+
| Total | Clock Enable | Synchronous | Asynchronous |
+-------+--------------+-------------+--------------+
| 0     |            _ |           - |            - |
| 0     |            _ |           - |          Set |
| 0     |            _ |           - |        Reset |
| 0     |            _ |         Set |            - |
| 0     |            _ |       Reset |            - |
| 0     |          Yes |           - |            - |
| 982   |          Yes |           - |          Set |
| 34295 |          Yes |           - |        Reset |
| 33    |          Yes |         Set |            - |
| 45600 |          Yes |       Reset |            - |
+-------+--------------+-------------+--------------+

2. Memory
---------

+-------------------+-------+-------+-----------+-------+
|     Site Type     |  Used | Fixed | Available | Util% |
[... 91 lines cut ...]
| RAMD64E  |   140 |  Distributed Memory |
| IBUF     |   134 |                  IO |
| RAMB36E1 |    93 |        Block Memory |
| RAMD32   |    72 |  Distributed Memory |
| FDSE     |    33 |        Flop & Latch |
| DSP48E1  |    33 |    Block Arithmetic |
| RAMS32   |    24 |  Distributed Memory |
| RAMB18E1 |    15 |        Block Memory |
| BUFG     |    12 |               Clock |
| SRL16E   |     8 |  Distributed Memory |
+----------+-------+---------------------+

8. Black Boxes
--------------

+----------+------+
| Ref Name | Used |
+----------+------+

9. Instantiated Netlists
------------------------

+----------+------+
| Ref Name | Used |
+----------+------+