nvdla/hw #91: HW architecture

ratnayak, 2018-01-24:
intergrator manual states: “nv_large” which has 2048 INT8 MAC’s"
HW architecture states for large nvlda implementation: " Increasing the Atomic-C and Atomic-K to 64/16 increases NVDLA performance to a maximum of 2K operations every cycle;"
and also states for large implementation: Atomic - C sizing = 64, Atomic - K sizing = 16

as 64x16=1024 could you pls explain the dependency between 2048 MACs and 64x16. Thanks.

jwise, 2018-01-25:
I think I answered somebody recently to say that INT8 performance is the same as INT16 performance -- in fact, I think this was you in #88 .  Upon further reading, this is, apparently, not true, and I was apparently mistaken: the effective Atomic-K size is doubled when running in INT8 mode.  Sorry for the error, and for the resultant confusion!  I'll also go update #88 to make sure that anybody who comes along afterwards is not confused.

ratnayak, 2018-01-30:
Thanks. thats clarifies my question.

closed 2018-01-30 by ratnayak