nvdla/hw #92: Running in Tools other than VCS

aarthym, 2018-01-25:
Hi, Wondering whether it is possible to run it in other Verilog Compiler, like INCISIVE? I am having trouble because of the PLI functions.  Are the PLI function definitions are included as the part of repo? I am not able to locate them 

Thank you!

jwise, 2018-01-26:
We don't officially support any tools other than VCS or Verilator, which include all the PLI functions that NVDLA uses.  Reports are that the current `nvdlav1` branch works in Xilinx's simulator, as well, but we don't officially support it.  If you have patches that allow NVDLA to run in Incisive, we'd be happy to accept contributions!

ShashankVM, 2020-07-16:
Is there a possibility of migrating to SystemVerilog?